NSPManager/NSPanelManager #1: Master buttons for ceiling lighst and table lights become desynchronized

tpanajott, 2023-03-06:
The master buttons (the two left ones) can switch state without the underlying C++ code handling the events.

Solution: Disable button handling in the panel and let the C++ code dictate the state of the buttons.

cablesandcoffee, 2023-03-07:
Hmm, we're talking about the ceiling and table lights buttons right? In Nextion Display they're "Dual State" buttons which can have a value of 0 or 1. They also have parameters called "picc" and "picc2". If the button "value" is 0 the picture shown is the value of "picc". If the button value is 1 the picture shown is the value of "picc2". I don't think there is a way of disabling the value changing when someone is pressing the button. It will just toggle between the different states. If it is 0 it becomes 1 if someone presses the button and the other way around. You could have the same picture for "value" 1 and 0 set in "picc" and "picc2" but then you would have to change the values of picc and picc2 when someone pressed the button? Or am I missing something/not understanding?

tpanajott, 2023-03-07:
Maybe it would be possible to implement a check that runs every, say, 5 seconds that would set the current "correct" value of the display. It seems like it shouldn't be needed but if that is what it takes I guess that would be a "fix"

cablesandcoffee, 2023-03-07:
Fixed with hotspots in commit: a88f164a604e842fcada6fa2bed370ab70665d9a

tpanajott, 2023-03-25:
This issue is back as of commit 23e5ef6

tpanajott, 2023-03-25:
Fixed again in commit b2d908452c4d08bdc1f30833667328f23ef76216

closed 2023-03-25 by tpanajott