NSPManager/NSPanelManager #2: All Lights mode + change Kelvin turn lights ON

cablesandcoffee, 2023-03-24:
Current behaviour:
Panel in All lights mode. No lights on. Changing Kelvin turns all lights on even those that does not exist.

Expected behaviour:
Only lights that are already on should receive the kelvin value.

tpanajott, 2023-03-25:
One possible solution to this might be to wait sending the new kelvin value for lights until they are turned on. The value could be stored in the MQTTManager as a new object that represents the light and when turning the light on, sending all the values stored for the light to OpenHAB/Home Assistant. However, this would require sending both kelvin and dimming level at the same time and would make the response times longer, especially over protocols that are not made to transfer lots of data, for example Zigbee of 433MHz.

The question then becomes in what order to send commands? Send new light level first or send kelvin and then lightlevel?

cablesandcoffee, 2023-03-25:
Yes I agree. I think that the current "Room kelvin" (if in Room lights mode) or current All lights kelvin (if in All lights mode)  value should be sent out when a light goes from 0. That way we sohlve a common problem that people are trying to solve them selves or that they've only been dreaming about having fixed. :) . I have a rule in Openhab that checks if a light goes from 0 and if so a kelvin command is sent some milliseconds later. It works good on my setup and I have a lot of light. As long as there is a small delay I think the zigbee control can handle it. 

I think sending light level first and then kelvin is a good solution. That is what Im used to at least and it works real good. :)

tpanajott, 2023-03-26:
Confirmed working in Home Assistant and OpenHAB.

closed 2023-03-26 by tpanajott