NSPManager/NSPanelManager #3: All lights mode, no lights ON + changing Brightness slider

cablesandcoffee, 2023-03-24:
Current behaviour:
Panel in All lights mode
No lights on
Changing Brightness dimmer results in all lights turning on even lights that does not exist.

Expected behaviour:
Only lights that are currently on should change.
If no lights are on only lights that exist should turn on.

tpanajott, 2023-03-25:
What exactly is meant by "only lights that exist should turn on"? Are lights that are not defined in the web interface turned on in the real world?

cablesandcoffee, 2023-03-25:
Haha no sorry. Right now if i have a room where only table lights exist. The ceiling light icon also becomes yellow in this case.
Also if I have a room with no lights, the display still paints ceiling and table icons yellow.

tpanajott, 2023-03-25:
But that is because the panel is in "all lights"-mode, it displays the average for all the lights

cablesandcoffee, 2023-03-25:
Oh! Yes you are right. My mind got lost again. Sorry. Solved! :)

tpanajott, 2023-03-25:
This issue is back as of commit 23e5ef6

closed 2023-03-25 by tpanajott