NSPManager/NSPanelManager #6: Add light Switch support

cablesandcoffee, 2023-03-25:
**Current behaviour**
Currently an Openhab user is able to add a light that is of type Switch but Openhab throws an error when NSPM is sending a percent value to the switch and the light is neither turned on or off.

Error from Openhab log: 
`[hab.core.io.websocket.EventWebSocket] - Failed to process deserialized event '{"type": "ItemCommandEvent", "topic": "openhab/items/Outlet2LightsMirror_Brightness/command", "payload": "{\"type\":\"Percent\",\"value\":\"0\"}", "source": "WebSocketNSPanelManager"}': Incompatible datatype, rejected.`

**Expected Behaviour**
If brightness slider > 0 a light of switch type should turn on
If brightness slider = 0 light of switch type shiuld turn off

Solution:
For switches the payload sent on websocket must look like this:
"payload": "{\"type\":\"OnOff\",\"value\":\"ON\"}"

tpanajott, 2023-03-26:
This should work as of commit 7728d939cb8d5ea881a9acd4778a080784f17fe4 but it requires further testing/verification

cablesandcoffee, 2023-03-26:
I can see that the add light window has been updated but it does not seem to work perfectly just yet. It looks like this:

[image]

The switch item is presented in Dimming , Color and RGB drop downs but not in Switch drop down. Get error when trying to add light as well:

[image]

tpanajott, 2023-03-26:
Try clearing the browser cache by holding shift while pressing the "reload"-button.

tpanajott, 2023-03-26:
Tested this issue using the latest code and can add lights as switches without it crashing. This requires more information. Can you please provide the whole page for the error?

cablesandcoffee, 2023-03-27:
Works after clearing cache. No issues with adding lights so everything seems solved.

closed 2023-03-27 by cablesandcoffee