NSPManager/NSPanelManager #9: All lights mode not settings lights to the same brightness

cablesandcoffee, 2023-03-25:
**Conditions:**
Panel in All Lights mode. Lights are on in several rooms and they are set to different percentages.

**Current behavior:**
When using the brightness slider all Lights that are on does not get the value where the user released the slider. Different values are passed to the lights and very soon after release the slider jumps to a new value.

**Expected behavior:**
All lights that are on should get the exact same value. The value where the slider was released.

tpanajott, 2023-03-25:
This might be because of trying to send to many MQTT messages, rework has begun to only send one MQTT-messages from the panel in order to reduce this risk of this

cablesandcoffee, 2023-03-27:
Can't see any issues any more. Closing!

closed 2023-03-27 by cablesandcoffee