NSPManager/NSPanelManager #17: Ability to "wake up" display from users home automation system

cablesandcoffee, 2023-03-28:
User should be able to wake the display up from HA/OH.

For example if a user has movement sensors they might want the display for that room to wake when someone enters the room.
I guess this should be done by having a "Wake Up" topic for each display that user can send a command to from HA/OH.

tpanajott, 2023-03-28:
That would be the simplest solution, yes. Simply a topic that if you send "1" it will turn on and "0" it will turn off. This would also make it simple to implement as a "switch" in a users home automation software and the panel could back feed information and update the topic based in state as well. If the panel goes to sleep, send a "0", if it wakes up, send a "1".

tpanajott, 2023-04-05:
Use the topics nspanel/<panel name>/screen_state or /screen_cmd to recieve/send "1" or "0"

closed 2023-04-05 by tpanajott

cablesandcoffee, 2023-04-05:
Cool! Works like a charm!