NSPManager/NSPanelManager #21: Implement "Display name" for lights

tpanajott, 2023-04-02:
Implement a "display name" for lights that is shown in the web interface and on the NSPanel. This is so that the light "id" isn't displayed as it can be long and contain a lot of information that is not needed when shown.

cablesandcoffee, 2023-04-04:
Very good. Noticed that as well. We should check how many characters that fit on one room page row and have that as max characters.

cablesandcoffee, 2023-04-06:
Max characters should be 15 for the "Display Name". Max is set to 17 for the Device text components on Room page so we can also add the two blank spaces.

tpanajott, 2023-04-07:
All lights currently added will have to be removed and readded for them to start working again. However, I'm guessing you may want to do that any way to set custom names for them

closed 2023-04-07 by tpanajott

cablesandcoffee, 2023-04-07:
Woohoo! Can't wait to try it!