NSPManager/NSPanelManager #66: Implement screen fields on screensaver

frontfoto, 2023-07-29:
Would be nice to have easy way to get for example the current temperature on screensaver where the clock is.
If it will possible some weather info, ex. Today Raining between 10-14 degrees

tpanajott, 2023-07-31:
Pinging @cablesandcoffee. Perhaps we should implement the most common things that people may want to display on the screensaver page but also give users the ability to send text to a specific field on the screensaver? That would mean that users could show any text they wanted simply by sending it over MQTT though it has limitation in that it cannot display icons and so on.

Roukie686868, 2023-07-31:
The MQTT message content on the screen saver would be great. That would allow me to not only do local temp but also a notification or something.

cablesandcoffee, 2023-08-06:
Yes I think a lot of people would like to have weather info on the screensaver. I like the idea with a custom text field to send notifications for example, that should be easy to fix. About the weather I think we have to do the entire weather icons thing where the display has preloaded icons for all possible weathers like a lot of other projects have. I think that is what is expected by the users. Don't now the best way to get the data to the manager though if it should come from HA/OH or an API connection straight to a weather site and the user just tells which city they're in?

cablesandcoffee, 2024-01-27:
I'm closing this one since most of this is implemented in the new mqtt-manager-rewrite. I will open a new one for the specific ability to send a user specified "notification message" to the screen.

closed 2024-01-27 by cablesandcoffee