NSPManager/NSPanelManager #67: IP address on boot screen.

Roukie686868, 2023-07-30:
While the unit boots up and connects to the manager it would be nice to have the IP address shown shortly.

tpanajott, 2023-07-30:
Perhaps @cablesandcoffee could update the TFT to include a secondary info-box on the boot screen?

Roukie686868, 2023-07-31:
Would be great. Due to some circumstances I have more than one DHCP server in my network. On a reboot of a panel it often gets a new IP address. The NSPanelManager has seems to have issue with that as it takes some time to change the redirect (IP address) to the panel page. When the boot screen shows the latest IP I can get there manually.

tpanajott, 2023-07-31:
I don't think the manager would automatically update the IP-addreas of the panel but you would have to make a manual refresh or the web page but I would think you already tried that.

Roukie686868, 2023-07-31:
I think the panel went back to the old IP address and so the link worked again. I got to find the time to fix my multiple DHCP problem. Went all works well there is very little need to go the the panel web page anyway.

tpanajott, 2023-07-31:
Yhea, the whole idea is that you change settings on the panel once when you set it up and then don't have to touch it again. That doesn't excuse this bug though, it should still work.

cablesandcoffee, 2023-08-06:
Yes will add this to my todo list!

cablesandcoffee, 2024-01-14:
Did some work on the GUI and happened to remember this one so I've just added a text box on the bootscreen to be able to print out panel ip address. Updated tftdefines files with info on the textbox.

cablesandcoffee, 2024-01-27:
This works, closing! :D

closed 2024-01-27 by cablesandcoffee