NSPManager/NSPanelManager #68: Request: Update to support NSPanel Pro

mbspears, 2023-07-30:
This is a great project.  Nice work.  Would be great if it also worked on NSPanel Pro.

tpanajott, 2023-07-30:
Hi! Thank you for your nice words but the NSPanel pro is not currently in the plan. It runs Android and not the ESP32-chip the firmware is written for. In the feature, it might be possible to emulate the interface as a webpage, but, as I said, there are other higher priority tasks at the moment.

mbspears, 2023-07-30:
Well keep up the great work and thanks for answering.  I didn't realize that there was such a difference.  Might be an incentive for me to get a NSPanel and follow your project!

closed 2023-07-30 by mbspears

velijv, 2023-07-30:
[Shameless self-promotion to close this topic about Pro ](https://community.home-assistant.io/t/nspanel-pro-ui-design-with-lovelace/541186)- run anything you want in it, design in lovelace, easy as pie.