NSPManager/NSPanelManager #70: Rename panel loses some connection

frontfoto, 2023-07-31:
When I change the name on the device in the manager the nspanel reboots but then shows offline. Even though it's reachable with the ip and can be controlled. If I remove the panel from manager it reboots and then it connects and shows online even though I need to configure it again.

tpanajott, 2023-07-31:
Where did you rename the panel and how did you do it? It has to be renamed in the manager and then the panel itself has to be rebooted for it to take affect.

frontfoto, 2023-07-31:
Unsure if I renamed it in manager or on the panels individual page. On one of this pages. I might have been on panels individual page. It didn't create a new panel in software and the link to the panels page still worked even though the manager listed it as offline.

mainmind83, 2023-09-06:
May be related to: https://github.com/NSPManager/NSPanelManager/issues/79

tpanajott, 2024-08-31:
This issue should never happen with the new upcoming stable release of the rewrite of the manager. Closing this issue.

closed 2024-08-31 by tpanajott