NSPManager/NSPanelManager #72: New error in H A 2023.8 related to NS Panel

aaronouthier, 2023-08-03:
[image]

I should probably note that the VM with my NSPanel manager is down at the moment.

tpanajott, 2023-08-03:
As written in the commit message, I think this fix will make the integration worse for OpenHAB users but this is something that has been directed to do from Home Assistant so any MQTT auto-discovery entity in OpenHAB using the Home Assistant compatibility would have this issue. Maybe @cablesandcoffee could have a look and check if there is something that could be done to easily make the entities (I guess "items" in case of OpenHAB) be clearly tied to the panel device?

tpanajott, 2023-08-05:
This will be fixed in the new release. Closing as no further action will be taken upon this issue but another enhancement issue may be created if there is an issue in OpenHAB with this change.

closed 2023-08-05 by tpanajott

cablesandcoffee, 2023-08-06:
I will have a look an see how Openhab behaves with the fix. It sound like someone has to update the Openhab HA autodiscovery function as well. Don't now how much time they put into that but will try and read and see how it works. One solution could be to help Openhab users with a script that creates a nice looking .thing file which they can add to their Openhab folder. I will do some thinking....