NSPManager/NSPanelManager #79: "Unknow status" on NSPanel-EU

mainmind83, 2023-09-06:
It's possible connecto to NSPanel IP, manager show dBm, ºC, % CPU... but always show "unknow status", when trying send TFT file with "Update screen" not show progress...

[image]

The manual does not show help on the options "[upload new tft file](https://github.com/NSPManager/NSPanelManager/tree/main/docker/web/nspanelmanager)", "upload new data file"...

tpanajott, 2023-09-06:
Can you make sure the panel has connected to mqtt? There is unfortunately no indicator on the panel or in the web interface so you will have to check your MQTT broker.

mainmind83, 2023-09-06:
Ohh, when you change name of panel, not updated name on MQTT, I delete with MQTT Explorer and from Manager and add again, now its working ;)

tpanajott, 2023-09-06:
Oh, interesting bug. I'm currently reworking the handshake and backend of the container so this procedure will change in the future so hopefully this bug will not show up again.

mainmind83, 2023-09-06:
When the name of the panel is changed, the name of the associated HA entities does not change either.

tpanajott, 2023-09-06:
That's strange as it should send the updated name to HA but it might be that HA is not updating the already registered entity. Will have to check that.

tpanajott, 2024-04-26:
This is fixed in the latest beta release and in the upcoming stable based on the beta. Will close.

closed 2024-04-26 by tpanajott