NSPManager/NSPanelManager #82: [Bug] NSPanelManager v0.0.1442 - MQTT password

mainmind83, 2023-09-06:
Accessing the IP address of the panel, if any change is made and the "save" button is clicked, when the page is reloaded in the browser the MQTT password has been lost in the text box, it is filled with a very string long (probably the hash? ;D)

[image]

mainmind83, 2024-09-01:
The [bug remains](https://github.com/NSPManager/NSPanelManager/issues/173) in version 0.1.150

jamesmulcahy, 2026-10-25:
Backlog sweep: I did a batch pass over the open issues with Claude, checking each one against previous fixes and the current `devel` code.

This was v1 firmware. In the v2 firmware web UI the stored password is never sent back to the page. Leaving the field blank keeps the existing password, and there's a separate checkbox to clear it.

cc @tpanajott: I think this one can be closed.

closed 2026-10-25 by tpanajott