NSPManager/NSPanelManager #84: [Enhancement] Home Assistant Integration

mainmind83, 2023-09-06:
1.- Add "show in lateral panel" on HA addon

[image]

2.- Is it possible to read the rooms and scenes directly from HA?

tpanajott, 2023-09-06:
The scenes is planned for integration, have not thought about rooms.

tpanajott, 2023-09-16:
I don't seem to find any options for the addon that says it would enable it to be visible in the menu per the documentation: https://developers.home-assistant.io/docs/add-ons/configuration/

Do you happen to know of an open source addon we could take a look at and see how they have done it?

mainmind83, 2023-09-16:
ESPHome, Frigate, HACS, Studio Code Server, Zigbee2mqtt... allow to add to lateral panel from addon config ;)

tpanajott, 2023-09-16:
Thanks, will take a look when I can.

mainmind83, 2023-09-16:
Maybe it can be of help::

- https://developers.home-assistant.io/blog/2018/06/01/071-custom-panels?_highlight=panel#improved-custom-panel-support
- https://smarthomescene.com/guides/sidebar-shortcuts-customizing-the-ha-sidebar/

tpanajott, 2024-08-31:
Unfortunetly this will be a big enough change that it will now be done before publishing the new "beta" as stable. I've done some digging and what has to be done is the following:
1. Change from using raw ports in config.yaml and start using ingress.
2. In using ingress, all paths need to be updated as simply pointing the browser to gather a file from /static/css/main.css will try and load it from Home Assistant and not the addon. The path indiciated in `X-Ingress-Path` has to be appended to all paths.

At least now we know what needs to be done to get it running we can start to actually give this issue some love and hopefully have it resolved in the next release.

mainmind83, 2024-12-08:
Thanks, great work 🙃