NSPManager/NSPanelManager #86: [Bug] Restarting HA/Panel

mainmind83, 2023-09-06:
If Home Assistant is restarted and the panel is still active, when HA starts, the panel appears as connected in the Manager but the screen shows "registering to manager"

tpanajott, 2023-09-06:
Yhea, the manager only looks at the mqtt status topic for the panel and if it's state is "online", that will however occur as long as it's connected to mqtt and might be before it has registered to the manager. As said in another post, this handshake between the manager and the panel is being reworked.

mainmind83, 2023-09-06:
If the IP of the panel changes, it is not updated either

tpanajott, 2023-09-06:
> If the IP of the panel changes, it is not updated either

What do you mean not updated? Not automatically updated on the page without a refresh or that it doesn't even show the correct IP after you reload the page?

mainmind83, 2023-09-06:
Not show the correct/new IP of Panel...

With new IP when this screen appear, trying access show "page Not found" on browser too

tpanajott, 2023-09-06:
I'm assuming you are running it as a HA add-on?

mainmind83, 2023-09-06:
yes, HA add-on with NSpanel v1.0 Protocol (by default)

tpanajott, 2023-09-06:
The protocol is just for when updating the screen, v1.2 is faster :)

mainmind83, 2023-09-07:
> The protocol is just for when updating the screen, v1.2 is faster :)

So it should be enabled by default when loading firmware ;D

tpanajott, 2023-09-07:
> So it should be enabled by default when loading firmware ;D

Yes, and I thought it was. What browser were you using at first setup?

mainmind83, 2023-09-08:
Chrome

tpanajott, 2024-08-31:
Closing this issue as I can't reproduce it. Please test with the current beta software or wait for it to go as the new stable and see if you still have the same issue. If so, please reopen this issue and send over the logs.

closed 2024-08-31 by tpanajott