NSPManager/NSPanelManager #87: https:// gives CSRF verification failed.

h3n3b0sch, 2023-09-06:
I have NSPanelManager installed in Proxmox in a LXC container using an Alpine-docker container.

There is reverse proxy via Nginx Proxy Manager with a Let's Encrypt SSL certificate. This SSL gives problems with contacting the panels and when saving anything to the panel it gives a Forbidden (403) error:

[image]

[image]

When opening the NSPanel Manager via http:// everything looks just fine:

[image]

Anyone got a solution to get https:// working?

h3n3

P.S. Great job btw, this NSPanelManager!

aaronouthier, 2023-09-06:
When accessing the panel manager from inside your home network, I would suggest using the IP address of your nspanelmanager directly from within the individual panel's UI, over http.

If you worry about security and feel that you need to access via HTTPS, then you'll need to run your own DNS server in your LAN/Home network.

Research "Split-brain DNS".

The issue you are likely having is, the domain name doesn't exist inside your LAN, so traffic from your individual Panel(s) is/are going out to the Internet, and then coming back in via your port forwarding on your router. Most home routers don't handle that situation properly.

You can try running OpenWRT/DD-WRT, or PFSense/OpnSense if you have supported hardware. The easier method might be to run something like PiHole. If you don't want the blocking, just remove the blocklist, and run the DNS Server.

aaronouthier, 2023-09-06:
Doing a whois lookup, the domain name opdethuisserver.nl doesn't seem to exist. Neither does nspmanager.opdethuisserver.nl . However, that could be a problem on my side. Are you sure those names exist?

h3n3b0sch, 2023-09-06:
> Doing a whois lookup, the domain name opdethuisserver.nl doesn't seem to exist. Neither does nspmanager.opdethuisserver.nl . However, that could be a problem on my side. Are you sure those names exist?

Yes, it is my own.

I have pi-hole running here, maybe i'll look in to that.

Still wondering... the settings of the panel are in IP-addresses not the domainname, why can't it get the status of the panel?

tpanajott, 2023-09-06:
> Yes, it is my own.
>
> I have pi-hole running here, maybe i'll look in to that.
>
> Still wondering... the settings of the panel are in IP-addresses not the domainname, why can't it get the status of the [...]

The actual status of the panel is gathered over MQTT and is then fetched through a websocket connection to the container when displaying the web interface. Have you allowed port 8001, which is a websocket, to the container?

h3n3b0sch, 2023-09-07:
> The actual status of the panel is gathered over MQTT and is then fetched through a websocket connection to the container when displaying the web interface. Have you allowed port 8001, which is a websocket, to the container?

It is allowed, without the https it connects.

h3n3b0sch, 2023-09-07:
For now I have disabled the outside FQDN because I realized it is unsafe without having a login page, eventhough it is not possible to change anything.
I have no understanding of Django (yet) but if I can find the energie and time I will try to implement a login page.

For now I will put this issue on a hold until then.

tpanajott, 2023-09-07:
You could try and set the trusted origins for the CSRF validation, example: CSRF_TRUSTED_ORIGINS = ['https://front.bluemix.net']

This should be added to the setting.py file in the container. If it works it's easy enough to add it as an environment variable for the container.

tpanajott, 2023-09-07:
> For now I have disabled the outside FQDN because I realized it is unsafe without having a login page, eventhough it is not possible to change anything.
> I have no understanding of Django (yet) but if [...]

It wouldn't necessarily be unsafe if the FQDN is only available in your network and the address is firewalled from the internet/untreated networks, in that case it's just a helpful name instead of remembering the IP.

jamesmulcahy, 2026-10-25:
Backlog sweep: I did a batch pass over the open issues with Claude, checking each one against previous fixes and the current `devel` code.

This should work on current versions. I tested it behind Nginx Proxy Manager with a Let's Encrypt certificate, forwarding plain HTTP to the container. Saving settings and panel settings both work, and the panel status loads.

The 403 came from Django's CSRF check. On a form POST, the browser's `Origin` header (`https://your.domain`) has to match the scheme and host Django thinks it is serving. In 2023 the container only saw plain HTTP from the proxy, so the two didn't match. The current container uses the `X-Forwarded-Proto: https` header that NPM sends by default (Caddy and Traefik send it too), so Django now knows the original request was HTTPS.

If you still get a 403 behind a proxy, check that:
- the proxy sends `X-Forwarded-Proto` (Apache `mod_proxy`, for example, doesn't by default), and
- it passes the original `Host` header through rather than rewriting it to the container's address.

The 403 page with `DEBUG` on shows the reason, e.g. "Origin checking failed - https://… does not match any trusted origins".

The panel status problem in the original report was the WebSocket on a separate port (8001). It now goes through the same port as the web UI at `/websocket`. In NPM, turn on "Websockets Support" for the proxy host.

@h3n3b0sch, could you retest on the current version?

cc @tpanajott: worth closing if there's no reply. If people turn up with proxies that can't send the header, we could add an environment variable for extra `CSRF_TRUSTED_ORIGINS`, as you suggested in the thread.

tpanajott, 2026-10-28:
Closing issue as no replay has been received. Thanks for testing @jamesmulcahy. Please reopen if issue is still present.

closed 2026-10-28 by tpanajott