NSPManager/NSPanelManager #89: Soft Button Size

mgbolts, 2023-09-13:
G'day Guys

May I suggest an alternate button pattern option.  The buttons on the screen are very small and require concentration in order to trigger the correct one.  

Many rooms dont require so many buttons, it would be nice to have a simpler 4 or 6 button screen for some rooms. ie. larger buttons.    

Thanks Mark

cablesandcoffee, 2024-01-27:
I think it is a good suggestion. I made the design with what I thought was the maximum manageable number of buttons. I think it works but I do press on the wrong buttons sometimes, and yes you have to be quite precise. :) I've been thinking about a solution where the room page adapt the size of the buttons depending of the amount of items the user has added to that room. That would require more images to be added to the gui file. For example 4 button image, 6 button image, 8 button image, 10 button image 12 button image. I would also require some logic from the manager to change this picture when entering all the different rooms. This increases the files size of the tft file and has to be weighed against all the other page pictures we will be adding with all the increased functionality. I will keep thinking about this and see what @tpanajott says about this when we start adding more pages to project.

tpanajott, 2025-04-23:
Hi, this is implemented in the upcoming beta as of issue #182. Will close this for now.

closed 2025-04-23 by tpanajott