NSPManager/NSPanelManager #91: Stuck at setting up AP

stleusc, 2023-09-16:
I currently have panels that are already flashed with custom firmware. That allows me to easily switch from Tasmota to ESPHome. And of course I don't need direct HW access anymore. So my thought was to simply do the initial FW upgrade through the UI (in my case ESPHome). Thankfully ESPHome complained about the firmware and did not flash it initially.

I realized that firmware.bin likely was what I needed. So I flashed that file without any problem. The panel rebooted and the AP showed up.
I connected to it and tried to access http://192.168.1.1 and the panel answers.... UNFORTUNATLY it answers with code 404, page not found.

Any advice?

stleusc, 2023-09-16:
When I try for example http://192.168.1.1/index.html I still get the 404 but this time it also returns content:

```
Path/File not found!
```

aaronouthier, 2023-09-16:
Are you trying to access with chrome browser on Android, by any chance?
There is a known issue, according to the user guide.

aaronouthier, 2023-09-16:
If all else fails, you can try reflashing from scratch, but first wait and see what the devs have to say.
Also, make sure you stay connected to the correct AP. Some devices will auto-disconnect if there is no Internet, and reconnect to a known working AP.

stleusc, 2023-09-16:
I think what I did does not work with this project. Which is a shame! I think the 'merged' file has extra stuff at the beginning so it is rejected by ESPHome and Tasmota flasher.
HOWEVER, the firmware.bin file seems to be missing the content of index.html (which I think is in data_file.bin).
Meaning, I flashed the FW but not the data! It sure would be nice to have a file that is compatible with Panels that are already flashed.

In any case now I am trying to flash again via USB and that does not work either....

I am using Espressif ESP32 DOWNLOADER TOOL and when I say start it just switches to SYNC and that is it.... I don't see any progress!?

aaronouthier, 2023-09-16:
You should be flashing /docker/web/nspmanager/merged_flash.bin to address 0x0 .

aaronouthier, 2023-09-16:
Check the documentation, here:
https://github.com/NSPManager/NSPanelManager/blob/main/docs/tex/manual.pdf

stleusc, 2023-09-16:
> You should be flashing /docker/web/nspmanager/merged_flash.bin to address 0x0 .

Yes, but the tool does nothing. But I remember I think I need to ground that one extra pin. Need to find instructions for that.

aaronouthier, 2023-09-16:
https://youtu.be/sCrdiCzxMOQ?si=cg0SCus0b0zd-ki5
Starts about 4:30

aaronouthier, 2023-09-16:
Basically, the display has 2 5v and 2 ground pins, and you can take one of the ground and connect direct to gpio 0

stleusc, 2023-09-16:
> Basically, the display has 2 5v and 2 ground pins, and you can take one of the ground and connect direct to gpio 0

Yeah, just remembered that too. Thanks for your help!!!

stleusc, 2023-09-16:
It's flashed....

I will still leave this open for the devs to look at.
What I think would be useful is the merged_flash.bin file without the first 0x1000 bytes.
They are 0xFF anyways and I don't think they are essential. Then this 'new file' could be flashed directly from ESPHome/Tasmota panels. They check for the magic byte and this would work then.

Unsure why the devs added the first 0x1000 as 0xFF. Likely one could remove those and use ESPFlasher starting at 0x1000 and it would work as well.

Anyways, I guess what I am saying is, it would be nice to have a file that can be flashed onto devices already running ESPHome/Tasmota. Since this project is rather new, there are MANY panels like that in the field. And those people would not have to open the panel to flash via USB.

aaronouthier, 2023-09-16:
You clearly know more about it than I.

Take care.

tpanajott, 2023-09-16:
> It's flashed....
>
> I will still leave this open for the devs to look at.
> What I think would be useful is the merged_flash.bin file without the first 0x1000 bytes.
> They are 0xFF anyways and I don't [...]

Glad you got it working and you are correct, the firmware.bin is only half of the puzzle. I seem to remember that you can flash our merged file via the esphome website, I know @cablesandcoffee did it in this video: https://youtu.be/dXKqnuqbTgk?si=ErEK5gUt0EPGtptu

We could have a look at stripping out the first parts of the merged file as those are essentially boot manager and stuff that shouldn't change regardless of firmware.

tpanajott, 2024-08-31:
Will close issue as cleanup as there hasn't been an update in a long time and it was fixed.

closed 2024-08-31 by tpanajott