NSPManager/NSPanelManager #93: Repository/Add-on Not Listing In Home Assistant Add-on Store Even Though Repository Is In List

BrettEBowman, 2023-09-23:
I've added this github repository to the Home Assistant Add-on Store:
[image]
However, in the Add-on Store screen, neither the Repository or the Add-on are being displayed:
[image]

tpanajott, 2023-09-24:
It seems that the automatic change for version number messed up the configuration file that HA is looking at when listing add-on details. It should be fixed now (manually). Have a look again and check if you can find it.

BrettEBowman, 2023-09-24:
> It seems that the automatic change for version number messed up the configuration file that HA is looking at when listing add-on details. It should be fixed now (manually). Have a look again and check if you can find it.

Unfortunately, no. I even tried removing the repository and re-adding it. Maybe it is a timing thing. I can try again a little later.

tpanajott, 2023-09-24:
Had another look and saw that even more things where messed up. Fixed them so hopefully it should work next time you check.

BrettEBowman, 2023-09-24:
> Had another look and saw that even more things where messed up. Fixed them so hopefully it should work next time you check.

Yes. That got it to show up and install.

Minor thing: on the web gui, the lower right corner is still showing "Version: 0.1.20.1"

tpanajott, 2023-09-24:
Hm, yhea, will have to take a look at that. There are some issues with the auto-updating of version numbers that need to be looked at.

tpanajott, 2023-10-01:
This has been resolved, closing.

closed 2023-10-01 by tpanajott