NSPManager/NSPanelManager #94: [Enhancement] capability to connect to Athom Homey Pro

eelco2k, 2023-10-01:
It would be super nice to have an connection between the NSPanelManager  and Athom Homey Pro 
(NSPanel MQTT Manager  <-- --> and the Athom Homey Pro  MQTT-Hub app)

i think this is the "easiest" method of implementation because the MQTT Hub app broadcasts all devices to the MQTT broker)
the app can be seen here @ https://homey.app/nl-nl/app/nl.hdg.mqtt/MQTT-Hub/

It uses the Homie Convention v3.0.1  (https://homieiot.github.io/specification/spec-core-v3_0_1/) and also a Home Assistant Discovery 

You're already building a super nice product! Keep going!

eelco2k, 2023-10-09:
after digging more in the code i think for implementing Homey you need this python library:

https://github.com/rkokkelk/python-athom-api

tpanajott, 2023-10-09:
> after digging more in the code i think for implementing Homey you need this python library:
> 
> https://github.com/rkokkelk/python-athom-api
> 

Seems like a wrapper library for their web API, shouldn't be needed from what I understand from the link in your first post. Good reference for implementation of our own API though.

eelco2k, 2023-10-09:
aah okay, i was confused because the home-assistant and openhab uses API key to connect and in both python files you make a websocket connection to set the values like this:

```
 msg = {
            "id": next_id,
            "type": "call_service",
            "domain": "light",
            "service": "turn_on",
            "service_data": {
                "hs_color": [
                    color_hue,
                    color_saturation
                ],
                "brightness_pct": light_level
            },
            "target": {
                "entity_id": entity_name
            }
        }
        send_message(json.dumps(msg))
        ```
        
      but it is true, that you could get / set all settings for homey via only mqtt :-) 👍