NSPManager/NSPanelManager #96: NSPanels on custom IP subnet does not work

eelco2k, 2023-10-08:
When my docker container runs in 10.0.0.0/24 range, and my NSPANELS for example in 10.0.30.0/24 range, the NSPanels won't popup in the NSPanelManager list.

logs i get:

```
08/Oct/2023 20:24:08] "GET /api/get_nspanels_warnings HTTP/1.1" 500 122963
Internal Server Error: /api/get_nspanel_config
ERROR:django.request:Internal Server Error: /api/get_nspanel_config
[08/Oct/2023 20:24:14] "GET /api/get_nspanel_config?mac=C4:DD:57:AA:11:11 HTTP/1.1" 500 0
```

aaronouthier, 2023-10-08:
I have a similar setup, and it DOES work if setup correctly.

You’ll need to give more info on your network setup though in order to track down your issue.

Personally, I use vlans, and had to setup routing between subnets, as well as open the correct firewall ports.

I would check if you’re able to get MQTT messages from the nspanel to your MQTT broker.

Go to the IP of your actual panel in your browser, and make sure you have your MQTT settings set correctly, then, check if the MQTT broker is receiving the messages. Manager should discover the panel if they are both communicating with the MQTT broker.

eelco2k, 2023-10-08:
@aaronouthier i also use VLANs. It is connected to MQTT as i see /nspanel/{NSPANELNAME}/command , log , r1_state, r2_state, screen_state, status, status_report, temperature_state. Topics.

also the topics: /nspanel/config/reload = 1
and: /nspanel/mqttmanager/command = "{"mac_origin":"C4:DD:57:11:11:11","method":"set","attribute":"brightness","brightness":0,"entity_ids":[]}"

So it has WiFi and also both can connect to MQTT, but i think something goes wrong with the websocket connection

eelco2k, 2023-10-08:
Perhaps it has something to do with the Docker networking service... i'm using Docker version 20.10.3, build 55f0773
and the container runs on the bridge network driver

eelco2k, 2023-10-08:
when i try from a computer connecting to the websocket  i can connect and get these messages:
```
{"type": "status", "payload": {"id": 1, "mac": "C4:DD:57:11:11:11", "name": "NSPANEL1", "is_us_panel": "False", "address": "172.17.0.1", "state": "online", "rssi": -68, "heap_used_pct": 42, "temperature": 27.42412758, "ip": "10.0.0.43", "warnings": ""}}
{"type": "mqttmanager_status", "mqtt": {"connected": true}, "home_assistant": {"configured": false, "connected": false, "auth_ok": false}, "openhab": {"configured": false, "connected": false}}
```

aaronouthier, 2023-10-08:
Double-check that your router/firewall has explicit rules allowing:
1. You to access your NSpanel from your computer/phone/tablet from which you're trying to connect to the WebUI.
and
2. Your NSpanel to access your MQTT broker.

I have setup my PFSense firewall to allow traffic from my main subnet to my IoT subnet as a blanket rule, and from my IoT network to the specific IP of my HomeAssistant OS, which also runs my MQTT.

Since I'm running my MQTT broker within HAOS, I'm not sure what docker networking type it uses.

eelco2k, 2023-10-08:
Weirdly enough putting on same subnet gives me the same error. So VLAN is not the issue that the NSPANEL wont fully start and stays at "Registering to manager..." So i will close this issue for now.

closed 2023-10-08 by eelco2k