NSPManager/NSPanelManager #99: Screen Upload Fails - Home Assistant

KenGrant, 2023-10-11:
I have successfully flashed my NS Panel & have it configured and connected to WiFi. I am at the stage of upload the initial tft screen but the panel crashes after intially showing it was going to update. From the debug  info I get the error here:

ERROR | NSPanel.cpp:737 f(._updateTFTOTA) Will now restart.
-- | --
15:39:20 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 5
15:39:20 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 0
15:39:20 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 57
15:39:20 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 93
15:39:20 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 7d
15:39:20 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 32

Please let me know what information you may need to dainose the issue. So keen to get this working.

tpanajott, 2023-10-11:
Could you please just verify your upload baud rate? It's a known issue that it doesn set correctly to 115200. If that doesn't work, try 9600

KenGrant, 2023-10-11:
Hi Tim,

Firstly, amazing project thank you for making this available on Github. I am late to this or would have supported you with Kickstarter.

A bit of background on my setup – My HA is behind a Proxy Server so from HA to Open the Web UI I need to use the local IP address to gain to load the GUI. I don’t this this is an issue as the Panel shows up as connected and the Firmware update seems to tun ok (I tried this in case it was an issue with Firmware).

You mention changing the baud rate , I have tried this at both rates and also using the Legacy protocol. Here is the screenshot just before it reboots.
[IMG_7223.jpg]

Please let me know what additional information might help.

Look forward to hearing from you.

Ken

tpanajott, 2023-10-11:
Hm, it seems you attached an image in a mail reply but that doesn't seem to have worked. Could you upload it via the GitHub interface? Thanks!

KenGrant, 2023-10-11:
Here is the screenshot just beforet the Panel reboots:-
[image]
Thanks

KenGrant, 2023-10-11:
I have tried all Baud rates as well as the legacy protocol - no luck.

tpanajott, 2023-10-11:
Hm, that's strange. Not seen that before. Usually it has a problem initiating the upload but as the progressbar is showing it has successfully initiated the update.

What does the HA addon log when you try to initiate the update? It should show many requests to something like /download_tft

KenGrant, 2023-10-11:
The log moves quickly - it fails shortly afte this - see earlier log dump....

INFO | NSPanel.cpp:636 f(._updateTFTOTA) Force restarting screen via power switch.
-- | --
17:50:23 | INFO | NSPanel.cpp:633 f(._updateTFTOTA) Will download TFT file from: http://192.168.1.52:8000/download_tft
17:50:23 | INFO | NSPanel.cpp:610 f(._updateTFTOTA) _updateTFTOTA Started.
17:50:23 | INFO | NSPanel.cpp:555 f(._taskUpdateTFTConfigOTA) Starting TFT update...
17:50:23 | INFO | InterfaceManager.cpp:30 f(.stop) Stopping interface manager.
17:50:12 | INFO | RoomManager.cpp:171 f(.loadAllRooms) Getting config for room 2
17:50:12 | DEBUG | RoomManager.cpp:32 f(.reloadCallback) Got reload command, reloading rooms.
17:50:12 | TRACE | MqttManager.cpp:115 f(._subscribeToTopic) Subscribing to topic nspanel/config/reload
17:50:12 | TRACE | MqttManager.cpp:115 f(._subscribeToTopic) Subscribing to topic nspanel/Den/r2_cmd
17:50:12 | TRACE | MqttManager.cpp:115 f(._subscribeToTopic) Subscribing to topic nspanel/Den/r1_cmd

If I try to dowload the tft link in a browser I get (I did think I might get a tft file to manually upload):

'NoneType' object has no attribute 'id'

Request Method: | GET
-- | --
http://192.168.1.52:8000/download_tft
4.1.7
AttributeError
'NoneType' object has no attribute 'id'
/usr/src/app/nspanelmanager/web/views.py, line 572, in download_tft
web.views.download_tft
/usr/local/bin/python
3.11.5
['/usr/src/app/nspanelmanager',  '/usr/local/lib/python311.zip',  '/usr/local/lib/python3.11',  '/usr/local/lib/python3.11/lib-dynload',  '/usr/local/lib/python3.11/site-packages']
Wed, 11 Oct 2023 17:53:34 +0000

tpanajott, 2023-10-11:
You will need to give it the Mac address as an argument. Try to look at the container/addon log when you issue a TFT upload command.

KenGrant, 2023-10-12:
I have found this in the container log - 

Not Found: /download_tft/name/A0:B7:65:54:CA:0C
WARNING:django.request:Not Found: /download_tft/name/A0:B7:65:54:CA:0C
[12/Oct/2023 06:14:05] "GET /download_tft/name/A0:B7:65:54:CA:0C HTTP/1.1" 404 10853

I have checked the MAC address and it's correct. I am not familiar with django but I do get a list of namesapce URL's with http://192.168.1.52:8000/index

Any thoughts? Is it worth deleting and re-installing in HA?

Thanks

KenGrant, 2023-10-12:
I have now moved to test a Docker container for the NSPanel Manager & MQTT - Panel apprears as before and Screen Update fails as before. For Info - I have purchased 2 Panels - both withn the last week - could there be a component change?

tpanajott, 2023-10-12:
> I have found this in the container log -
>
> Not Found: /download_tft/name/A0:B7:65:54:CA:0C
> WARNING:django.request:Not Found: /download_tft/name/A0:B7:65:54:CA:0C
> [12/Oct/2023 06:14:05] "GET [...]

That does not seem correct. The panel should send a request to /download_tft which should look at the IP-address the request is coming from in order to figure out whether to download the EU or US version of the file.

tpanajott, 2023-10-12:
Could you please enable debug logging and upload the log from the NSPanen when you try to upload a new TFT? It seems you added some logs in your original message for this issue but not all data in that table is present.

KenGrant, 2023-10-12:
Hi Tim,

I hace moved back onto HA - The live log in the Panel only has ten lines and it moves through quickly. I can't see a way of send you everything that get shown in the log as I attempt the screen update. - there is no scrolling through the history.

tpanajott, 2023-10-12:
> Hi Tim,
> 
> I have moved back onto HA - The live log in the Panel only has ten lines and it moves through quickly. I can't see a way of send you everything that get shown in the log as I attempt the screen update. - there is no scrolling through the history.
> 

In the settings in the manager you can configure how many lines it should display.

tpanajott, 2023-10-12:
Thanks!

It looks like the display is sending out some kind of data, though not quite sure what as I can't see a return code of 0x55 in the Nextion instruction set https://nextion.tech/instruction-set/

tpanajott, 2023-10-12:
Converting the hex-vales to ASCII you get some jibberish and then {HMI_Interface: 2}

This is probably a Sonoff TFT thing where they print the version or something. Not seen it before but something we should deal with. Will probably have a firmware ready for you to test later today where we clear the read buffer just before sending our command to flash. Or, simply wait longer after restarting the screen so that the existing clear fixes it.

Thanks for the logs!

tpanajott, 2023-10-12:
Hi!

Please try the attached firmware and data file.  You can either flash the panel via serial again using the attached merged-flash.bin or you can upload the firmware.bin and data_file.bin in the "Upload"-menu in the web interface and do a OTA update from there.

[temp.zip](https://github.com/NSPManager/NSPanelManager/files/12886071/temp.zip)

KenGrant, 2023-10-13:
Hi

Many thanks for this - I uploaded OTA both files (presume order not important) and there was still an error although it seems to have got a bit further. Let me if you think re-flashing might be worth trying. Here is the error log:

06:17:00 | INFO | main.cpp:129 f(.taskManageWifiAndMqtt) taskWiFiMqttHandler started!
-- | -- | --
06:17:00 | INFO | main.cpp:240 f(.setup) Starting tasks
06:17:00 | DEBUG | NSPanel.cpp:407 f(._taskProcessPanelOutput) Read type 88
06:17:00 | DEBUG | NSPanel.cpp:407 f(._taskProcessPanelOutput) Read type 0
06:17:00 | DEBUG | NSPanel.cpp:407 f(._taskProcessPanelOutput) Read type 88
06:17:00 | DEBUG | NSPanel.cpp:407 f(._taskProcessPanelOutput) Read type 0
06:17:00 | ERROR | main.cpp:235 f(.setup) Failed to initiate NSPanel.
06:17:00 | INFO | NSPanel.cpp:251 f(.init) NSPanel::init complete.
06:17:00 | INFO | NSPanel.cpp:345 f(._taskReadNSPanelData) Starting taskReadNSPanelData.
06:17:00 | INFO | NSPanel.cpp:312 f(._taskSendCommandQueue) Starting taskSendCommandQueue.
06:17:00 | INFO | NSPanel.cpp:237 f(.init) Trying to init NSPanel.
06:16:59 | WARNING | NSPanel.cpp:232 f(.init) No stored MD5 for TFT. Is the panel flashed with NSPanel Manager TFT?
06:16:59 | INFO | NSPanel.cpp:228 f(.init) Got reply from display: comok 2,30614-0,NX4832F035_011C,51,61744,17B13401ED9E1E38,16777216
06:16:59 | DEBUG | NSPanel.cpp:212 f(.init) Sending final connect to panel
06:16:59 | INFO | NSPanel.cpp:200 f(.init) Trying to connect to display.
06:16:59 | DEBUG | NSPanel.cpp:180 f(.init) Got text from panel:
06:16:59 | DEBUG | ButtonManager.cpp:110 f(._loop) Started ButtonManager _loop.
06:16:59 | INFO | main.cpp:231 f(.setup) Initializing NSPanel communication
06:16:41 | INFO | InterfaceManager.cpp:30 f(.stop) Stopping interface manager.

KenGrant, 2023-10-13:
Hi

Many thanks for this - I uploaded OTA both files (presume order not important) and there was still an error. However, I think I missed a step - I thought uploading the new data files & firmware would also run the update firmware on the panel (maybe not?). I am thinking the first uploads are to the Panel Manager in HA for firmware update manually run from each Panel - I guess this makes sense?

I have now re-run firmware update from the panel - the firmware update now completes but the screen now just says System Data ERROR in Portrait mode. Here is the error log and I'll attach a picture. (P.S. - Co-operation is all yours - many thanks!).

06:45:08 | INFO | main.cpp:129 f(.taskManageWifiAndMqtt) taskWiFiMqttHandler started!
-- | -- | --
06:45:08 | INFO | main.cpp:240 f(.setup) Starting tasks
06:45:08 | ERROR | main.cpp:235 f(.setup) Failed to initiate NSPanel.
06:45:08 | INFO | NSPanel.cpp:251 f(.init) NSPanel::init complete.
06:45:08 | INFO | NSPanel.cpp:345 f(._taskReadNSPanelData) Starting taskReadNSPanelData.
06:45:08 | INFO | NSPanel.cpp:312 f(._taskSendCommandQueue) Starting taskSendCommandQueue.
06:45:08 | INFO | NSPanel.cpp:237 f(.init) Trying to init NSPanel.
06:45:08 | INFO | NSPanel.cpp:228 f(.init) Got reply from display: comok 2,30614-0,NX4832F035_011C,72,61744,17B13401ED9E1E38,16777216
06:45:08 | DEBUG | NSPanel.cpp:212 f(.init) Sending final connect to panel
06:45:08 | INFO | NSPanel.cpp:200 f(.init) Trying to connect to display.
06:45:08 | DEBUG | NSPanel.cpp:180 f(.init) Got text from panel:
06:45:08 | DEBUG | ButtonManager.cpp:110 f(._loop) Started ButtonManager _loop.
06:45:08 | INFO | main.cpp:231 f(.setup) Initializing NSPanel communication
06:44:53 | DEBUG | NSPanel.cpp:407 f(._taskProcessPanelOutput) Read type 1a
06:44:53 | INFO | WebManager.cpp:261 f(._taskPerformOTAUpdate) Will restart in 5 seconds.
06:44:52 | INFO | WebManager.cpp:246 f(._taskPerformOTAUpdate) Successfully updated LittleFS.
06:44:52 | WARNING | WebManager.cpp:343 f(._update) OTA Done, will reboot!
06:44:52 | INFO | WebManager.cpp:341 f(._update) OTA Successful!
[image]

KenGrant, 2023-10-13:
Hi 
Update - I wondered if the issue related to not having any Devices setup in my Panel. I chose a lamp and re-uploaded the screen - It worked! 
Let know know if you would like me to try anything further. I'll also now check in with Discord (if I can find it). I am hoping there might be a write up on the enviroment/tools you are usng to achive this great work. I have covers (shutters) that I am planning to use the physical buttons for opening & closing for now - I did read you are planing on expanding from just lights?

Many thanks for help - my NSPanel (Manager) journey can start........Will send a cup of coffee to the project.

KenGrant, 2023-10-13:
Ok - Understood - perhaps just running screen update twice did the trick.

So far panel working just fine although just 1 device going to try scenes etc this afternoon- I'll raise another issue if I find anything.

Just to mention - Screensaver time out seems falcky & clock isn't showing - set to true on Panel & Global.

Lastly - what is your preference sending any Bugs, issues, feedback?

Many thanks

KenGrant, 2023-10-13:
Ah - Ignore Screen saver - brightness default was 0 - changed tp 50 - All good.

tpanajott, 2023-10-13:
It's probably best to send issues and enhancements on GitHub (as this one is). Regarding the clock though, that is flaky and has been reworked in the upcoming version with the new MQTTManager

KenGrant, 2023-10-13:
Excellent - so pleased to get started with this - thanks.

I'll close this now.

closed 2023-10-13 by KenGrant