NSPManager/NSPanelManager #101: Add Docker compose file?

aaronouthier, 2023-10-12:
What would it take to get a docker-compose.yml file created for this project?

I don't understand enough about json/yaml to create it myself. I can usually figure out enough to adapt an existing compose file to my needs, but currently there is nothing for me to adapt.

There is no way I'g going to type that docker run command every time, and the method I'm using now (cd docker; ./docker-build-and-run.sh) hasn't worked for the last week or so, due to a compilation error of some kind. I have pulled the latest source with git.

tpanajott, 2023-10-13:
Getting a docker-compose example up should be pretty quick. Though you should probably download the zip from the releases tab or download a specific release tag in git to have it working. Could you elaborate more on what error you are getting when trying to compile?

Also, what machine are you running this on?

aaronouthier, 2023-10-13:
x64 virtual machine. Debian 12 VM/docker host.

` => ERROR [7/7] RUN pip install -r requirements.txt                                                                                            192.0s
------                                                                                                                                                
 > [7/7] RUN pip install -r requirements.txt:                                                                                                         
2.797 Collecting asgiref==3.6.0 (from -r requirements.txt (line 1))                                                                                   
2.985   Downloading asgiref-3.6.0-py3-none-any.whl (23 kB)                                                                                            
3.067 Collecting certifi==2022.12.7 (from -r requirements.txt (line 2))                                                                               
3.079   Downloading certifi-2022.12.7-py3-none-any.whl (155 kB)
3.119      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 155.3/155.3 kB 4.1 MB/s eta 0:00:00
3.353 Collecting charset-normalizer==3.0.1 (from -r requirements.txt (line 3))
3.363   Downloading charset_normalizer-3.0.1-py3-none-any.whl (45 kB)
3.381      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 45.5/45.5 kB 3.1 MB/s eta 0:00:00
3.610 Collecting Django==4.1.7 (from -r requirements.txt (line 4))
3.620   Downloading Django-4.1.7-py3-none-any.whl (8.1 MB)
5.024      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 8.1/8.1 MB 5.8 MB/s eta 0:00:00
5.129 Collecting django-environ==0.10.0 (from -r requirements.txt (line 5))
5.146   Downloading django_environ-0.10.0-py2.py3-none-any.whl (19 kB)
5.200 Collecting django-mathfilters==1.0.0 (from -r requirements.txt (line 6))
5.209   Downloading django_mathfilters-1.0.0-py3-none-any.whl (5.9 kB)
5.326 Collecting django-ranged-response==0.2.0 (from -r requirements.txt (line 7))
5.336   Downloading django-ranged-response-0.2.0.tar.gz (3.0 kB)
5.351   Preparing metadata (setup.py): started
5.802   Preparing metadata (setup.py): finished with status 'done'
6.156 Collecting gevent==22.10.2 (from -r requirements.txt (line 8))
6.173   Downloading gevent-22.10.2.tar.gz (6.6 MB)
6.930      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 6.6/6.6 MB 8.7 MB/s eta 0:00:00
7.719   Installing build dependencies: started
15.02   Installing build dependencies: finished with status 'done'
15.02   Getting requirements to build wheel: started
21.09   Getting requirements to build wheel: finished with status 'done'
21.10   Installing backend dependencies: started
22.20   Installing backend dependencies: finished with status 'done'
22.20   Preparing metadata (pyproject.toml): started
23.36   Preparing metadata (pyproject.toml): finished with status 'done'
23.77 Collecting greenlet==2.0.2 (from -r requirements.txt (line 9))
23.78   Downloading greenlet-2.0.2.tar.gz (164 kB)
23.82      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 165.0/165.0 kB 4.1 MB/s eta 0:00:00
23.87   Preparing metadata (setup.py): started
24.15   Preparing metadata (setup.py): finished with status 'done'
24.20 Collecting idna==3.4 (from -r requirements.txt (line 10))
24.21   Downloading idna-3.4-py3-none-any.whl (61 kB)
24.24      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 61.5/61.5 kB 2.6 MB/s eta 0:00:00
24.29 Collecting paho-mqtt==1.6.1 (from -r requirements.txt (line 11))
24.30   Downloading paho-mqtt-1.6.1.tar.gz (99 kB)
24.34      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 99.4/99.4 kB 2.9 MB/s eta 0:00:00
24.36   Preparing metadata (setup.py): started
24.59   Preparing metadata (setup.py): finished with status 'done'
24.69 Collecting pip-upgrade==0.0.6 (from -r requirements.txt (line 12))
24.71   Downloading pip-upgrade-0.0.6.zip (6.3 kB)
24.73   Preparing metadata (setup.py): started
24.94   Preparing metadata (setup.py): finished with status 'done'
25.20 Collecting psutil==5.9.4 (from -r requirements.txt (line 13))
25.21   Downloading psutil-5.9.4-cp36-abi3-manylinux_2_12_x86_64.manylinux2010_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl (280 kB)
25.27      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 280.2/280.2 kB 5.2 MB/s eta 0:00:00
25.39 Collecting pytz==2023.3 (from -r requirements.txt (line 14))
25.42   Downloading pytz-2023.3-py2.py3-none-any.whl (502 kB)
25.53      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 502.3/502.3 kB 4.5 MB/s eta 0:00:00
25.64 Collecting requests==2.28.2 (from -r requirements.txt (line 15))
25.65   Downloading requests-2.28.2-py3-none-any.whl (62 kB)
25.68      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 62.8/62.8 kB 2.4 MB/s eta 0:00:00
25.72 Collecting sqlparse==0.4.3 (from -r requirements.txt (line 16))
25.73   Downloading sqlparse-0.4.3-py3-none-any.whl (42 kB)
25.75      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 42.8/42.8 kB 2.6 MB/s eta 0:00:00
25.85 Collecting urllib3==1.26.14 (from -r requirements.txt (line 17))
25.87   Downloading urllib3-1.26.14-py2.py3-none-any.whl (140 kB)
25.90      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 140.6/140.6 kB 4.2 MB/s eta 0:00:00
25.99 Collecting websocket-client==1.6.1 (from -r requirements.txt (line 18))
25.99   Obtaining dependency information for websocket-client==1.6.1 from https://files.pythonhosted.org/packages/d3/a3/63e9329c8cc9be6153e919e17d0ef5b60d537fed78564872951b95bcc17c/websocket_client-1.6.1-py3-none-any.whl.metadata
26.00   Downloading websocket_client-1.6.1-py3-none-any.whl.metadata (7.6 kB)
[... 186 lines cut ...]
[... cut ...]
191.4 Successfully built django-ranged-response gevent paho-mqtt pip-upgrade websockets zope.interface
191.4 Failed to build greenlet
191.4 ERROR: Could not build wheels for greenlet, which is required to install pyproject.toml-based projects
------
Dockerfile:15
--------------------
  13 |     WORKDIR /usr/src/app
  14 |     COPY web/ /usr/src/app/
  15 | >>> RUN pip install -r requirements.txt
  16 |     
  17 |     EXPOSE 8000
--------------------
ERROR: failed to solve: process "/bin/sh -c pip install -r requirements.txt" did not complete successfully: exit code: 1`

tpanajott, 2023-10-13:
Strange. Thanks for the logs, will try to look into it later. Regarding using a specific release you can have a look here: https://stackoverflow.com/a/792027

aaronouthier, 2023-10-13:
I wonder if I need to be logged-in as root in order to succeed? Currently, I am running as a member of the docker group. I have also tried downloading again (via git clone) to a different folder name, but get stuck in the same place.

I need to 'git' back to sleep now. It's 2 AM. I got up to hit the head, and checked my email on my way back to bed...

tpanajott, 2023-10-13:
I've now added a docker-compose example under the advanced setup in the manual.

closed 2023-10-13 by tpanajott