NSPManager/NSPanelManager #102: Screen update fails - maybe I missed something

gon0, 2023-10-13:
I just bought 2 NSPanels and have successfully added both of them to the NSPanelManager according to the .pdf-manual. It worked all fine until the screen update.
This file https://github.com/NSPManager/NSPanelManager/blob/main/HMI%20files/nspmanager.tft is uploaded as tft-file.

This is how it looks in Overview:
[image]

This is seen on the Panel during update:
[image]
After this, it shows the Sonoff-Logo again and after a short while, it shows the standard Sonoff-screen.

This is what the Debug output from the NSPanel shows:

```

17:28:10 | DEBUG | NSPanel.cpp:407 f(._taskProcessPanelOutput) Read type 55
17:27:10 | INFO | InterfaceManager.cpp:135 f(._taskProcessMqttMessages) Started _taskProcessMqttMessages.
17:27:10 | INFO | InterfaceManager.cpp:120 f(._taskLoadConfigAndInit) Config initialized. Closing taskLoadConfigAndInit
17:27:10 | DEBUG | InterfaceManager.cpp:117 f(._taskLoadConfigAndInit) Not attaching MQTT  clock callback is panel is confiugred to now show clock on screensaver.
17:27:10 | DEBUG | NSPanel.cpp:407 f(._taskProcessPanelOutput) Read type 55
17:27:08 | INFO | main.cpp:72 f(.registerToNSPanelManager) Registered to manager at: http://192.168.122.222:8000/api/register_nspanel
17:27:08 | ERROR | RoomManager.cpp:26 f(.reloadCallback) Received reload command within 5 seconds of a reload. Ignoring command.
17:27:08 | INFO | WebManager.cpp:39 f(.init) Starting web server.
17:27:08 | INFO | main.cpp:153 f(.taskManageWifiAndMqtt) Gateway:    192.168.122.1
17:27:08 | INFO | main.cpp:152 f(.taskManageWifiAndMqtt) Netmask:    255.255.255.0
17:27:08 | INFO | main.cpp:151 f(.taskManageWifiAndMqtt) IP Address: 192.168.122.234
17:27:08 | INFO | main.cpp:148 f(.taskManageWifiAndMqtt) Connected to WiFi Heizung
17:27:08 | INFO | MqttManager.cpp:216 f(._connect) Connected to MQTT server 192.168.122.40
17:27:08 | INFO | MqttManager.cpp:205 f(._connect) Connecting to MQTT server 192.168.122.40
17:27:08 | ERROR | main.cpp:137 f(.taskManageWifiAndMqtt) WiFi not connected!
17:27:08 | INFO | main.cpp:129 f(.taskManageWifiAndMqtt) taskWiFiMqttHandler started!
17:27:08 | INFO | main.cpp:240 f(.setup) Starting tasks
17:27:08 | ERROR | main.cpp:235 f(.setup) Failed to initiate NSPanel.
17:27:08 | INFO | NSPanel.cpp:251 f(.init) NSPanel::init complete.
17:27:08 | INFO | NSPanel.cpp:345 f(._taskReadNSPanelData) Starting taskReadNSPanelData.
17:27:08 | INFO | NSPanel.cpp:312 f(._taskSendCommandQueue) Starting taskSendCommandQueue.
17:27:08 | INFO | NSPanel.cpp:237 f(.init) Trying to init NSPanel.
17:27:08 | WARNING | NSPanel.cpp:232 f(.init) No stored MD5 for TFT. Is the panel flashed with NSPanel Manager TFT?
17:27:08 | INFO | NSPanel.cpp:228 f(.init) Got reply from display:
17:27:07 | DEBUG | NSPanel.cpp:212 f(.init) Sending final connect to panel
17:27:07 | INFO | NSPanel.cpp:200 f(.init) Trying to connect to display.
17:27:07 | DEBUG | NSPanel.cpp:180 f(.init) Got text from panel:
17:27:07 | DEBUG | ButtonManager.cpp:110 f(._loop) Started ButtonManager _loop.
17:27:07 | INFO | main.cpp:231 f(.setup) Initializing NSPanel communication
17:26:57 | ERROR | NSPanel.cpp:737 f(._updateTFTOTA) Will now restart.
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 0
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 57
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 93
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 7d
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 32
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 3a
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 22
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 65
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 63
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 61
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 66
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 72
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 65
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 74
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 6e
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 69
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 5f
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 49
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 4d
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 48
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 22
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 7b
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 0
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 13
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 87
17:26:57 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) aa
17:26:57 | ERROR | NSPanel.cpp:732 f(._updateTFTOTA) Failed to init upload to TFT. Got return code: 55
17:26:57 | DEBUG | NSPanel.cpp:723 f(._updateTFTOTA) Waiting for panel reponse
17:26:56 | INFO | NSPanel.cpp:705 f(._updateTFTOTA) Starting upload using v1.2 protocol.
17:26:56 | DEBUG | NSPanel.cpp:700 f(._updateTFTOTA) Will start TFT upload, TFT file size: 2766880
17:26:54 | DEBUG | NSPanel.cpp:665 f(._updateTFTOTA) Got comok: comok 2,30614-0,NX4832F035_011C,51,61744,51B03401539E1760,16777
17:26:54 | DEBUG | NSPanel.cpp:654 f(._updateTFTOTA) Sending connect to panel
17:26:48 | INFO | NSPanel.cpp:636 f(._updateTFTOTA) Force restarting screen via power switch.
17:26:48 | INFO | NSPanel.cpp:633 f(._updateTFTOTA) Will download TFT file from: http://192.168.122.222:8000/download_tft
17:26:48 | INFO | NSPanel.cpp:610 f(._updateTFTOTA) _updateTFTOTA Started.
17:26:48 | INFO | NSPanel.cpp:555 f(._taskUpdateTFTConfigOTA) Starting TFT update...
17:26:48 | INFO | InterfaceManager.cpp:30 f(.stop) Stopping interface manager.
```

When I access "http://192.168.122.222:8000/download_tft" via Firefox browser I receive:

[image]

What I already tried:

- switching between Latest ("v1.2") and Legacy (v1.0) 
- switching between different Baud rates 115200, 9600, 19200, 921600

I have to admit, that I was lost in the manual at the tft-flashing-part:
[image]

Is there something I have missed?

tpanajott, 2023-10-13:
Probably not, we just saw a similar situation from someone who bought new panels. It seems the new sonoff TFT has som additional stuff in it but it's been fixed. See issue #99.

We will probably publish a new release in a few hours that will fix it. Just hang tight.

Good error reporting though, you gave a lot of information.

gon0, 2023-10-13:
Wow, you are quick!

Sure, I am happy to hear you already have a solution.

Thank you :)

tpanajott, 2023-10-13:
New release just published. You will have to wait up to 1 hour for it to update for all platforms. Then it should work for you though as the manual states, it can be quite finicky so give it a try or two.

gon0, 2023-10-13:
Thank you for the quick fix.

I am not sure, what you mean by being updated on all platforms. I forgot to mention, that I am using the NSPanelManager as a Home Assistant Plugin. Here it shows version  0.1.23 (which was published 2 weeks ago).

I looked for a new .bin firmware file to flash and a new .tft-file, but since I could not find a new file, I am not sure if I looked correctly. For me this is not an urgent issue, so I'll just patiently wait :)

gon0, 2023-10-14:
I have no updated NSPanelManager to 0.1.25 via HA Add-Ons

After that, I uploaded the new merged-flash.bin to the NSPanelManager.

Since the .tft file was not changed on github, I did not update the .tft-file.

This is what the panel showed during update:
[image]

This is the DEBUG-output when updating the screen via v1.2 protocol @ 115200 baud:

```
14:00:21 | DEBUG | NSPanel.cpp:407 f(._taskProcessPanelOutput) Read type 55
13:57:37 | DEBUG | NSPanel.cpp:407 f(._taskProcessPanelOutput) Read type 55
13:56:37 | INFO | InterfaceManager.cpp:135 f(._taskProcessMqttMessages) Started _taskProcessMqttMessages.
13:56:37 | INFO | InterfaceManager.cpp:120 f(._taskLoadConfigAndInit) Config initialized. Closing taskLoadConfigAndInit
13:56:37 | DEBUG | InterfaceManager.cpp:117 f(._taskLoadConfigAndInit) Not attaching MQTT  clock callback is panel is confiugred to now show clock on screensaver.
13:56:37 | DEBUG | NSPanel.cpp:407 f(._taskProcessPanelOutput) Read type 55
13:56:35 | INFO | main.cpp:72 f(.registerToNSPanelManager) Registered to manager at: http://192.168.122.222:8000/api/register_nspanel
13:56:35 | ERROR | RoomManager.cpp:26 f(.reloadCallback) Received reload command within 5 seconds of a reload. Ignoring command.
13:56:35 | INFO | WebManager.cpp:39 f(.init) Starting web server.
13:56:35 | INFO | main.cpp:153 f(.taskManageWifiAndMqtt) Gateway:    192.168.122.1
13:56:35 | INFO | main.cpp:152 f(.taskManageWifiAndMqtt) Netmask:    255.255.255.0
13:56:35 | INFO | main.cpp:151 f(.taskManageWifiAndMqtt) IP Address: 192.168.122.234
13:56:35 | INFO | main.cpp:148 f(.taskManageWifiAndMqtt) Connected to WiFi Heizung
13:56:35 | INFO | MqttManager.cpp:216 f(._connect) Connected to MQTT server 192.168.122.40
13:56:35 | INFO | MqttManager.cpp:205 f(._connect) Connecting to MQTT server 192.168.122.40
13:56:35 | ERROR | main.cpp:137 f(.taskManageWifiAndMqtt) WiFi not connected!
13:56:35 | INFO | main.cpp:129 f(.taskManageWifiAndMqtt) taskWiFiMqttHandler started!
13:56:35 | INFO | main.cpp:240 f(.setup) Starting tasks
13:56:35 | ERROR | main.cpp:235 f(.setup) Failed to initiate NSPanel.
13:56:35 | INFO | NSPanel.cpp:251 f(.init) NSPanel::init complete.
13:56:35 | INFO | NSPanel.cpp:345 f(._taskReadNSPanelData) Starting taskReadNSPanelData.
13:56:35 | INFO | NSPanel.cpp:312 f(._taskSendCommandQueue) Starting taskSendCommandQueue.
13:56:35 | INFO | NSPanel.cpp:237 f(.init) Trying to init NSPanel.
13:56:35 | WARNING | NSPanel.cpp:232 f(.init) No stored MD5 for TFT. Is the panel flashed with NSPanel Manager TFT?
13:56:35 | INFO | NSPanel.cpp:228 f(.init) Got reply from display:
13:56:34 | DEBUG | NSPanel.cpp:212 f(.init) Sending final connect to panel
13:56:34 | INFO | NSPanel.cpp:200 f(.init) Trying to connect to display.
13:56:34 | DEBUG | NSPanel.cpp:180 f(.init) Got text from panel:
13:56:34 | INFO | main.cpp:231 f(.setup) Initializing NSPanel communication
13:56:34 | DEBUG | ButtonManager.cpp:110 f(._loop) Started ButtonManager _loop.
13:56:24 | ERROR | NSPanel.cpp:737 f(._updateTFTOTA) Will now restart.
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 0
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 57
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 93
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 7d
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 32
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 3a
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 22
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 65
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 63
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 61
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 66
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 72
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 65
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 74
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 6e
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 69
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 5f
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 49
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 4d
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 48
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 22
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 7b
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 0
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 13
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 87
13:56:24 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) aa
13:56:24 | ERROR | NSPanel.cpp:732 f(._updateTFTOTA) Failed to init upload to TFT. Got return code: 55
[... 264 lines cut ...]
14:10:40 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 80
14:10:40 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 80
14:10:40 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 80
14:10:40 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 80
14:10:40 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 0
14:10:40 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 80
14:10:40 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 80
14:10:40 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 80
14:10:40 | INFO | NSPanel.cpp:734 f(._updateTFTOTA) 80
14:10:40 | ERROR | NSPanel.cpp:732 f(._updateTFTOTA) Failed to init upload to TFT. Got return code: 80
14:10:38 | DEBUG | NSPanel.cpp:723 f(._updateTFTOTA) Waiting for panel reponse
14:10:38 | INFO | NSPanel.cpp:705 f(._updateTFTOTA) Starting upload using v1.2 protocol.
14:10:38 | DEBUG | NSPanel.cpp:700 f(._updateTFTOTA) Will start TFT upload, TFT file size: 5124856
[... cut ...]
14:10:23 | DEBUG | NSPanel.cpp:654 f(._updateTFTOTA) Sending connect to panel
14:10:17 | INFO | NSPanel.cpp:636 f(._updateTFTOTA) Force restarting screen via power switch.
14:10:17 | INFO | NSPanel.cpp:633 f(._updateTFTOTA) Will download TFT file from: http://192.168.122.222:8000/download_tft
14:10:17 | INFO | NSPanel.cpp:610 f(._updateTFTOTA) _updateTFTOTA Started.
14:10:17 | INFO | NSPanel.cpp:555 f(._taskUpdateTFTConfigOTA) Starting TFT update...
14:10:17 | INFO | InterfaceManager.cpp:30 f(.stop) Stopping interface manager.
```

Is there is anything else I could try?

gon0, 2023-10-17:
By using the latest .tft and firmware file, updating the screen worked 👍 

Thanky for adapting so quick :)

closed 2023-10-17 by gon0