NSPManager/NSPanelManager #103: will it work on docker compose

66schoollane, 2023-10-19:
i looked in the manual and found this 
```
docker run -- name nspanelmanager -v / etc / timezone :/ etc / timezone : ro -v \
" $ ( pwd )/ web / nspanelmanager / db . sqlite3 " : " / usr / src / app / nspanelmanager / db . sqlite3 " \
" $ ( pwd )/ web / nspanelmanager / secret . key " : " / usr / src / app / nspanelmanager / secret . key " \
-d -p 8000:8000 -p 8001:8001 nspanelmanager
```
so from this & info from docker pull nspanelmanager/nspanelmanager-armv7:latest

how do i do a compose.yaml file, not sure if this start of one is remotely correct, as I need to add more info:

 nspanelmanager:
    image: nspanelmanager/nspanelmanager-armv7:latest
    container_name: nspanelmanager
    ports:
      - 8000:8000
      - 8001:8001
    volumes:
      - 
      - 
    environment:
    restart: always

thanks

tpanajott, 2023-10-19:
There is a section on using docker compose in the advanced section in the manual. Have a look and come back if you have any more questions.

tpanajott, 2023-10-19:
We had some problems with publishing the latest release so the updated PDF didn't get copied to the docker container. Check here instead: https://github.com/NSPManager/NSPanelManager/blob/main/docs/tex/manual.pdf

The link to the manual on the README page also points to the above file as that is the file that will be most up to date.

66schoollane, 2023-10-19:
@tpanajott Thanks

closed 2023-10-19 by 66schoollane