NSPManager/NSPanelManager #104: Not able to update screen

minhix, 2023-10-22:
Hi!
I'm facing a problem as updating screen on my NS Panel.
- On nspanels manager plugin (on homeassistant) it keeps showing "Updaing GUI 0%". 
- On device, it shows no progress
```
USART Update...
0
Size:
5124936
writing DATA...
```

Could someone help explaining what's wrong?

---
Setup:
- Install nspanel manager plugin v0.1.25 to Homeassistant (v2023.10.3)
- Following manual to install nspanel `merged-flash.bin` (v0.1.25) to the nspanel.
- Configure nspanel. It could be found on nspanl manager
- Notice: "GUI Update available". Start "UPdate screen", but no progress

minhix, 2023-10-22:
Figured out the problem is solved when the baud rate is set to default, I did it set it at all before. Wish this is mentioned in the manual

closed 2023-10-22 by minhix

tpanajott, 2023-10-23:
With the latest release some people have had problems flashing the TFT, including @cablesandcoffee. We are actively working on trying to figure out what the problem is as there is not obvious signs on what the issue is. It works perfectly for some people and not at all for others.

chakalaka91, 2023-11-09:
> Figured out the problem is solved when the baud rate is set to default, I did it set it at all before. Wish this is mentioned in the manual

What settings did you use? It's seems like I've the same issues, but changing the baud rates doesn't seem to improive something. I can reach the Panel itself via IP, but now the Panel Manager doesn't recognize the Panel anymore.

tpanajott, 2023-12-07:
> Same problem here, seems to be there regardless of baud rate... 

What protocol version are you using? V1.1 has a bug and doesn't work.