NSPManager/NSPanelManager #106: [Enhancement] (At least for HA), Display Friendly Name Instead of/In Addition To Entity ID

BrettEBowman, 2023-10-29:
I'm only using NSPanelManager with HA (with the HA Addon), so I don't know how it works with OpenHAB or standalone. With the HA add-on, when it lists lights (in the Rooms screens, the panel settings screens, etc. plus on the panel screen) just display the HA Entity ID ("light.xxxx") and not the HA Friendly Name. More and more, then Entity IDs are generic and include device identifiers (like MAC address) and don't necessarily identify the actual light. For instance, see "light.smart_dimmer_switch_std_touch_2_light" on this screen:
[image]
The Friendly Name for that device is "Deck Lights":
[image]

It would be great to at least have the Friendly Name displayed when selecting entities. And it would be preferable to then use the Friendly Name to display the light both in setting screens and on the panel.

lucasfeijo, 2023-10-31:
FYI you can go in edit and change that entity ID to a friendly name, it will remain working as intended. That said, I think it still should be fixed to use the friendly name automatically.

BrettEBowman, 2023-10-31:
@lucasfeijo Thanks for that info. I had discovered that I could go in there afterwards and make the name friendlier. Which is helpful. But it doesn't help with finding the correct device to add in the first place when only the obscure Entity IDs are listed.

tpanajott, 2023-12-25:
Thanks you for reporting this enhancement. This is now implemented in the upcoming release of the new MQTTManager. I will leave it open until we have release the new MQTTManager as stable.

closed 2024-09-01 by tpanajott