NSPManager/NSPanelManager #107: Weather and time functionality on Screensaver page

cablesandcoffee, 2023-11-01:
There are a two enhancements that are in the same area so I created this one to keep track of the progress.
Referenced issues: #105 #66 

**Goal of enhancement:**
- Implement time and weather data on screensaver for users that wants that.

**What I've done so far:**

- Updated Screensaver page so we can show following data:
1. Current Time, date, room temp (user choice but could be the sensor on the NSPanel)
2. Current Sunset/Sunrise time
3. Current weather - Icon, temp, temp maxmin, wind, rain
4. Forcast weather 5 days - Icon, temp, temp maxmin, wind, rain

- Updated tftdefines file
- Created general font mapping json file. (used to know which icons are mapped to which characters in the font file used in Nextion)
- Created weather_mapping file containing mapping for both HA and OH.

- My thoughts about where the data comes from?
HA users: NSPanelManager gets it from Home Assistant weather functionality in some form. Tim knows that better than me.
Openhab: Will require Openhab users to setup two http get requests from Openweathermap.com. We will provide instructions o how to do that. It's very easy. This will result in two Openhab Item strings containing JSON data data NSPanelManager can use. Current weather and forecasted weather for the five coming days. Openhab only really has one good weather binding and that one is created with Openweathermap. The binding does not give us the original json data though so therefore I propose using Openhabs HTTP binding and let the user get the JSON data from there. This will make it easier to implement against Openhab I think.

- NSPanelManger choices:

1. Background/No background on screensaver page. Choice is made in NSPanelManager which results in a variable being set in Nextion screen on start of the screen. Logic of setting / removing background is solved by code prepared by Nextion screen. 
2. Weather/No Weather. This should be a function that has to be activated by the users. If not activated the weather items should be set to not visible. Might be a good solution to do the same as the background choice (another variable is set in the Nextion screen and the hiding/showing is taken care of from there?)
3. Should not need to handle units, mm, celsius and so on. I'm thinking it should be known in the data that is received?

tpanajott, 2023-11-01:
This is a very nice feature to have but I would say that it's not useful to add it to the current version. This is something that would be implemented with the new MQTTManager.

cablesandcoffee, 2023-11-01:
Oh yes that's the plan of course. Maybe there's a way to create issues for different branches. I didn't think of that. 😊

tpanajott, 2024-01-14:
As of latest beta in MQTT Manager rewrite, maybe this could be closed?

cablesandcoffee, 2024-01-27:
Yes I agree, lets close it ! :)

closed 2024-01-27 by cablesandcoffee