NSPManager/NSPanelManager #109: Update screen stuck at 0%

chakalaka91, 2023-11-09:
I stumbled over the NSPanel and ESPHome a few weeks ago and immediately purchased one. In the last few days I've had so many failed attempts to install ESPHome, when I've discovered NSPManager. I was willing to give this a try.

The installation and flashing process worked just fine. I was able to flash the nspanel via esp web without any problems. Now when I try to update the screen it apparently doesn't do anything. I'd assume it has anything to do with my attempts to update the tft with the HA blueprint from blackymas, but I'm quite lost, what I could do to fix this. Blackymas Blueprint has a option to clear the tft with a blank file, maybe this could be an option to blank the tft and retry again?

tpanajott, 2023-11-09:
What TFT is already on the display shouldn't matter. You can try to change between baud 9600 or 115200 in the settings page on the panel and give it a try. If that doesn't work, please enable debug logging and send the logs over.

chakalaka91, 2023-11-09:
I've already tried several different baud settings, none works.

On debug level the log always repeats the same steps:

<html>
<body>

16:36:50 | INFO | NSPanel.cpp:636 f(._updateTFTOTA) Force restarting screen via power switch.
-- | -- | --
16:36:50 | INFO | NSPanel.cpp:633 f(._updateTFTOTA) Will download TFT file from: http://192.168.178.10:8000/download_tft
16:36:50 | INFO | NSPanel.cpp:610 f(._updateTFTOTA) _updateTFTOTA Started.
16:36:45 | ERROR | NSPanel.cpp:581 f(._taskUpdateTFTConfigOTA) Failed to update TFT. Will try again in 5 seconds.
16:36:44 | ERROR | NSPanel.cpp:742 f(._updateTFTOTA) Will try to upload anyway. If nothing happens, please reboot and try again.
16:36:44 | ERROR | NSPanel.cpp:737 f(._updateTFTOTA) Failed to init upload to TFT. Got return code: ff
16:36:44 | DEBUG | NSPanel.cpp:728 f(._updateTFTOTA) Waiting for panel reponse
16:36:44 | DEBUG | NSPanel.cpp:723 f(._updateTFTOTA) Sending TFT upload command: whmi-wri 6107436,1115200,1
16:36:44 | DEBUG | NSPanel.cpp:720 f(._updateTFTOTA) Doing one last serial2 clear.
16:36:44 | INFO | NSPanel.cpp:713 f(._updateTFTOTA) Starting upload using v1.1 protocol.

</body>
</html>

tpanajott, 2023-11-09:
It seems that the V1.1 is somewhat broken as of now (there is a PR pending review fixing it). Try with V1.2 protocol.

chakalaka91, 2023-11-09:
Weirdly enough the option the change the protocol somehow disappeared. 

[image]

chakalaka91, 2023-11-09:
So there's an interesting behaviour. The option for the protocol disappears when I don't enter the mqtt password, when doing a change on the NSPanel. However if I don't enter the password each time I'm doing a change, the nspanel appears in state offline in the nspanel manager, even though I can access it via ip directly. 

Now I did change the baud rate without entering the mqtt password and the option for the protocol reappeared. But now the nspanel manager shows the nspanel as offline :D 
[image]
[image]

With entering the mqtt password the nspanel is shown as online in the nspanel manager, but the protocol option disappeared. 
[image]
[image]

With the previous option for the protocol enabled, I've changed it to 1.2, but apparently it gets overwritten. When I try to upload the tft file it still uses the 1.1 protocol.

tpanajott, 2023-11-09:
Hm, that's sounds very strange. The online/offline part in the manager simply goes by MQTT. Given that the panel can't connect to mqtt without the password it would indeed be shown as offline.

I'll have to take a look regarding the field for protocol disappearing when you enter a password as that shouldn't happen and I've not seen that happening for anyone else.

chakalaka91, 2023-11-10:
It's still the case with the missing protocol part, but I've managed to update the screen now. I've tried several constellations with the baud rate, with mqtt password and without it. At some point it did save the protocol and I could update the screen which worked now.

Is it intentional, that you have to enter the mqtt password every time you make a change on the panel?

chakalaka91, 2023-11-10:
So I'm not entirely sure, what fixed the problem, but it definitly was a combination of making changes to the panel with the password, then without, changing the baud rate and changing the protocol. What worked for me is the default baud rate with 115200 and the protocol 1.2. Although this was the default setting from the beginning it wouldn't work just then. I've did some of the mentioned changes, after which I got it to work. The screen update then took about 3-5min.

closed 2023-11-10 by chakalaka91

tpanajott, 2023-11-13:
Glad you got it running. If you don't mind me asking, do you perhaps have a "-character in your password? I can see that messing things up.

chakalaka91, 2023-11-13:
@tpanajott: no, there's no " character. Though I do have a % and a # as special characters. The username has a - in between two names.

tpanajott, 2023-11-13:
Thanks for the info. Don't quite know why it wouldn't show up then.