NSPManager/NSPanelManager #110: Updating GUI 0% - The NSPanel reboots but the GUI doesn't update

turpinant, 2023-11-22:
Hello everyone,

First of all, a huge thanks to the developers of this project. It's the first time I'm trying to contribute to a github project, so please excuse me if raising an issue for my situation is not the way to go. Also, I am french, not a native english speaker, so please apologize for any mistakes :)

I succeded in flashing my NSPanel (not pro) EU version yesterday night. I can see it in my NSPanel dashboard in HA. I installed the HA addon, FYI.

I can currently see a yellow tooltip, telling me that there is an update for the GUI (I uploaded the gui.tft file found in docker/web/nspanelmanager).

I also updated the firmware, no issue here.

When I click on "update GUI", I can see some sort of progress bar, telling me "Updating GUI 0%". The screen reboots, show me the sonoff logo. On the NSPanel UI, I can see the device going offline, and rebooting with still the default TFT displayed on the screen.

I saw in the manual guide (PDF) that the GUI update process can be buggy, but I think I already tried something like 30 times. Still no success.
I found another tft file under HMI files called nspmanager.tft. Maybe I should try this tft file instead? I am currently at work, so I can't test it.

Anyway, I would love some guidance on what I am doing wrong, or a fix if this is a real know issue.
Thank you a lot for taking the time to read me.

tpanajott, 2023-11-22:
Hi, you are correct in raising an issue here. Regarding the TFT file, you don't need to upload any files manually.

What settings do you have in the page on the panel itself? Regarding baud and upload protocol.

turpinant, 2023-11-22:
Hi,
I will update my post tonight when I get back, because I unplugged the NSPanel this morning. 
But yesterday I left the baud empty, and also tried the default value, it still failed. I have no idea what settings I am running for the upload protocol. I'll send you these info, along with some screenshots tonight.

tpanajott, 2023-11-22:
Try baud 115200 or 9600 and protocol v1.2 as v1.1 has a bug currently.

turpinant, 2023-11-22:
Hi again, I was using protocol 1.2, so I left this unchanged. 
I immediately tried baud 9600, and the GUI update started properly, and is ongoing as I write this. Thank you for the useful tip ! I'll update this post if I encounter any issue during the update process.

turpinant, 2023-11-23:
Hi again,
just to confirm that the update of the GUI went smoothly yesterday night. It took approx 2 hours to complete I think, I didn't really count precisely, but it's done :)
On to customizing it now woohooo !
I guess I can close this issue.

Thanks again for the support and for this great addon.

tpanajott, 2023-11-23:
Glad you got it running! :)