NSPManager/NSPanelManager #111: [Feature Request] Mean value of temperature to be transmitted instead of noisy actual value

gon0, 2023-11-28:
The temperature sensor of the panel is not a precise instrument, since the data is quite noisy:
[image]

However, after adjusting the temperature offset (https://github.com/NSPManager/NSPanelManager/issues/88), the transmitted temperature value seems to fluctuate around the actual value.

It seems a temperature value is transmitted each 30 seconds:
[image]

Is there a chance to calculate a mean-value or exponential filter, before the temperature value is sent? I would propose to collect 30 temperature values within 30 seconds (1 value sampled per second) and create the mean value from it. Transmitted is only the mean value, to smooth out fluctuations (if this is not already done).

tpanajott, 2023-11-28:
Hey!

Thanks for your request and it's something I will definitely look into!

tpanajott, 2024-01-24:
This will be implemented on the new upcoming beta and forward. Currently implemented but will keep the issue open until it can be tested and verified buy someone more than just me.

tpanajott, 2024-04-26:
Fixed in latest beta and soon to be stable release.

closed 2024-04-26 by tpanajott