NSPManager/NSPanelManager #115: Uploading TFT fails, missing instructions

johnneerdael, 2024-01-03:
Hi updating the TFT fails and I keep gettings errors while trying to update, what's mostly frustating is that I cannot find information anywhere about which TFT file I should be uploading in the first place? Its also going ridiculously slow so it would really help if someone could provide some more detailed steps.

tpanajott, 2024-01-03:
Hi!

You do not need to choose a TFT file. Only thing you need to do is check the box that it is a US panel, in case it is (this is in the panel specific settings). When uploading it will be very slow. Even at the highest speed (baud rate, set in the web interface on the panel) it will take about 10 minutes.

johnneerdael, 2024-01-04:
Unfortunately even after 10m its not succeeding, it's to the point it has no valid TFT on it now and I'm not getting a resolution on what I could do to resolve it. I've read about many similar issues but none of the pointers I have received yet provided any resolution

tpanajott, 2024-01-04:
What happens on the screen of the panel? It should display progress for the upload. If this doesn't happen you can try to change the upload baud in the panels web interface to 9600 or 115200. Many people have had luck with those two.

tpanajott, 2024-01-30:
> I was FINALLY able to get the upload to work by setting the upload speed to 9600 (which resulted in the above error), followed by setting the "US Panel" (I am in the US), then setting the upload speed to the default and clicking the upload.  
> 
> In my setup, the upload speed was not set to anything at first. I had to go through that procedure to eliminate the error " Trying to add command to queue before a queue exists!" 
> 
> HTH 

Nice that you got it working and yes, switching around the baud rate is usually the fix. Switching between 115200 (default for sonoff firmware and ours) and 9600 (default for Nextion) seems to fix it.

tpanajott, 2024-01-30:
> I spoke too soon. The TFT upload works but then the Panel reports "System Data ERROR!"   I reloaded the merged-flash.bin file again, set the correct parameters from the resultant site (at 192.168.1.1), restarted the NSPanel, and ran "Update screen" again at 115200 baud. 
> 
> Is this error ( "System Data ERROR!")  a failure of the NSPanel TFT screen update or something else?  Thanks in advance.
>  

It means the firmware checksum was invalid or the upload of the TFT failed for some other reason. As I've written elsewhere the upload is really sensitive. I know @cablesandcoffee have previously had to turn off a lot of other WiFi devices in order for it to succeed, no matter the firmware.

To answer your question, just keep trying. As long as you see the upgrade progress bar you're on the right track.

tpanajott, 2024-04-26:
I'm going to close this issue as it hasn't seen any action in a while. Please reopen or open a new one if you still encounter issues.

closed 2024-04-26 by tpanajott