NSPManager/NSPanelManager #116: System Data ERROR on screen

youmansk, 2024-01-08:
The panel shows up in the list and is active. The screen just says "system Data ERROR" and that is all. I tried to upload the TFT again but not working. This seems like a common error across several nspanel frameworks. I know it is because the tft is corrupt but i do not know how to upload a good one.

youmansk, 2024-01-08:
I think i got it. I had to upload a nspmanager.tft via nexiton App and tf-rx tf-tx. took like 10 minutes but i think it worked. Now the manager is telling me the gui is out of date but that should be ok.

tpanajott, 2024-01-08:
Yhea, the TFT screen itself is really finicky with the update but usually you can get it going. If it doesn't work the next time, try changing the upload baudrate to 9600 (instead of default 115200). This has worked for a lot of people.

Regarding the TFT being out of date it should be fine. It's just saying that because the firmware hasn't registered a successful upload of a TFT and stored the current TFT file version. Only problem I can see is that when we release a new TFT in the feature you won't notice it. But just keep an eye out for strange behavior on the screen (images missing and so on) and you should be fine.

baclaeys, 2024-01-23:
I am having the same problem.  I am using 115200 for the baud rate.  I have a US NSPanel.  The file transfer will stop at a different time every time, but it got to 93% one time before it fails and goes to the screen that says "system Data ERROR.  Let me know if any logs would be helpful.

tpanajott, 2024-01-23:
> I am having the same problem. I am using 115200 for the baud rate. I have a US NSPanel. The file transfer will stop at a different time every time, but it got to 93% one time before it fails and goes to the screen that says "system Data ERROR. Let me know if any logs would be helpful.

Hm, strange. Usually what the problem is is that the update itself doesn't start, ie. the screen showing what percentage of upload is complete never shows. I know @cablesandcoffee has had lots of issues of flashes not succeeding when having lots of other things online on the WiFi when flashing though that mostly shows after the flashing has gone to 100% and it says something like "integrity error" (please correct me if I'm wrong). I would say that perhaps @cablesandcoffee would have better idea of what this could be?

baclaeys, 2024-01-23:
I will attempt another transfer, capture the logs and post them here.

aaronouthier, 2024-01-24:
I just purchased my 2nd NSPanel, US version.
There are differences in the markings on the back of the board (top of the board, if you're taking off the back, with the LCD down).

Is it possible that Sonoff or Nexion changed something subtle, either to improve something, or possibly to confound projects like yours?

In the attached photos, the old board is on the right side, new board on the left. I do note that the Hardware revision is the same, and I don't see any physical components changed.
[image]
[image]

aaronouthier, 2024-01-24:
Once my friend solders the pin headers on my USB to TTL adapter, I'll try to get a boot log and firmware dump, before flashing it.

baclaeys, 2024-01-24:
I got home tonight, and it was showing system Data ERROR".  When I started the gui update again, and after it rebooted it started around 93% compete and finished.  It was really strange, but it took the gui update.

tpanajott, 2024-01-24:
The GUI update is really finiky. Having a lot of WiFi/Bluetooth devices can really mess with it. I know @cablesandcoffee had lots of issues in the beginning and many times had to turn off lots of devices in order to get it working, regardless of firmware (ESPHome or nspanelmaneger).

Glad you got it running though!

cablesandcoffee, 2024-01-27:
Sorry for the late answer. Seems like it magically worked of some reason. Haven't seen that behavior, that it finishes as 93%, failes, then starts on 93% next time you try. But if happens again the logs are very important as Tim says. Currently my tft uploads works 100% of the time but I have had issues were the panels has gotten to 97-99 isch % but at that point the nspanelmanager is att 100% and finishes the upload. Data is clearly lost a long the way of some reason when that happens. On those occasions we have seen interesting log messages which have been of help and Tim has been able to resend specific portions of data if I remember correctly. I've used the internal log that exist for every panel in the nspanelmanager web ui and changed the number of log row settings as Tim suggested to be able to collect everything.

tpanajott, 2024-08-31:
We've not had any more activity in this issue for a long time and I will close it. This is a known issue that happens no matter the framework you use to update the TFT to the panel. This is because of noise on the data lines to/from the TFT and ESP32.

closed 2024-08-31 by tpanajott