NSPManager/NSPanelManager #123: max/min temp current weather Openhab

cablesandcoffee, 2024-01-14:
Openhab Accuweather

Max/min temp in top left box is always 0/0

Not sure if you have tried to put some data in there yet. I looked through the api-data and in the current weather data there is a section called "temperature summary" with subsection "past 6 hour summary". Iḿ thinking max/min from this section could be used for the max/min temp text box in upper left corner. What do you think?

tpanajott, 2024-01-16:
Please test again with latest release

cablesandcoffee, 2024-01-22:
Interesting. Does this have to do with the fact that a light can have a Switch "property" and a Brightness "property"?
In zigbee2mqtt for example I can turn a light OFF but that does not change the Brightness property, it will still stay the same. I can see that I get this issue if I use the switch/state property in zigbee2mqtt. I guess I've just been lucky cause I've only set up the Brightness property for my lights in Openhab. Since I only control the lights from Openhab I don't experience this problem. But for someone using Openhab that are actually using the switch entity that exists in zigbee2mqtt they would get the same problem. It would also be a problem if some user used physical zigbee switches on the walls for example. Doing that would result in a state change ON/OFF when turning lights ON/OFF, at least if using zigbee2mqtt.

cablesandcoffee, 2024-01-22:
Works!

closed 2024-01-22 by cablesandcoffee