NSPManager/NSPanelManager #125: [NSPanel Firmware] Room/Home page does not update when a brightness of 0 is received on any light

tpanajott, 2024-01-16:
If an on or off statement is sent to the panel via MQTT the switches on the "Room"-page does not update, neither does the home page.

cablesandcoffee, 2024-01-22:
Interesting. Does this have to do with the fact that a light can have a Switch "property" and a Brightness "property"?
In zigbee2mqtt for example I can turn a light OFF but that does not change the Brightness property, it will still stay the same. I can see that I get this issue if I use the switch/state property in zigbee2mqtt. I guess I've just been lucky cause I've only set up the Brightness property for my lights in Openhab. Since I only control the lights from Openhab I don't experience this problem. But for someone using Openhab that are actually using the switch entity that exists in zigbee2mqtt they would get the same problem. It would also be a problem if some user used physical zigbee switches on the walls for example. Doing that would result in a state change ON/OFF when turning lights ON/OFF, at least if using zigbee2mqtt.

tpanajott, 2024-01-24:
I see the same problem in HA where the state of the light is not really used, only if the type is set to "switch" and not "dimmable". I can see that the manager sends out the new brightness of the light as 0 over MQTT but the panel itself doesn't really do anything with the value. I think it is discarded as we do not wish for the slider to go to 0 but to remember the last average value when all the lights were shut off. Somewhere during the implementation of that this bug must have slipped in.

tpanajott, 2024-01-24:
Just fixed a bug in the new MQTTManager and cannot reproduce this. Will keep it open as it is still an issue in current stable release. Do I understand correctly that @cablesandcoffee have this problem with OpenHAB?

tpanajott, 2024-04-15:
This is fixed in HA but it does seem like OpenHAB doesn't seem like OpenHAB sends out a state update or even registers that a light has turned to 0. Perhaps we can add this somewhere in the manual and close this issue?

cablesandcoffee, 2024-04-27:
Hmm I personally don't have the problem because I don't use the zigbee2mqtt switch state and only control the lights from Openhab. But for people that use it, it would be a problem. If a setup uses physical zigbee switches for example, when pressing OFF button the only thing that changes in zigbee2mqtt is the Switch state, not brightness. I guess it's correct of Openhab to not update the brightness state to 0 when the switch state changes to OFF since the brightness state has not changed. Only the switch state has. That is how zigbee2mqtt works at least. 

But if the user would have this kind of setup and experience this as an issue it should work with a workaround where the user setups a transformation on the brightness item in openhab and checks the switch state, if switch state is OFF set brightness to 0. If turning the switch ON after that zigbee2mqtt will still set the brightness value to the last used brightness value. 

So yes I think we could put it in the manual and close this.

tpanajott, 2025-07-15:
This will be included in the next update of the documentation as a 'Known issue'. Will close.

closed 2025-07-15 by tpanajott