NSPManager/NSPanelManager #138: [enhancement] Remotly (via MQTT) change sceensaver.

Ellghe, 2024-03-23:
Basiacally do would I like to shift screen saver mode, having the weather and clock during daytime and only the clock during the night time. (When the alarm is on or off).

tpanajott, 2024-04-07:
This is now somewhat working though it seems  we need to update the TFT as well. Maybe @cablesandcoffee and chime in with some thoughts? If I change the ssScreensaver variable after the page has been initialized it seems I cant turn on/off the background. Is it possible to fix that within the TFT or do wee need to find an other solution?

cablesandcoffee, 2024-04-09:
correct, it only ran the code on first load. I've removed the first load function so it will instead run the code every time the page loads. so if the variable has changed the background will change when the page is loaded.

tpanajott, 2024-04-10:
Confirmed working with new TFT.

cablesandcoffee, 2024-04-10:
Nice! :)

closed 2024-04-10 by cablesandcoffee

Ellghe, 2024-04-10:
My idea was not to chagnge the background but to changer from clock date & weather to only clock!