NSPManager/NSPanelManager #139: Beta version  FW can't connect to server

niedi74, 2024-03-24:
beta version install   fw 0.1.13 ,  which is not able to load config from server .

on the panel no Manager address can be  add

tpanajott, 2024-03-24:
Yhea, known issue. I'm guessing you aren't a member on the Discord. We are currently testing a fix that will hopefully go out today which should solve the issue.

niedi74, 2024-03-24:
ok, thx 

im not on discrod

closed 2024-03-24 by niedi74

tpanajott, 2024-03-24:
Okay, we put out a PSA there yesterday regarding this. You will unfortunately have to reflash the panels with the new firmware once we publish the update. As the panels aren't connecting to the manager you can't update them OTA unfortunately.

niedi74, 2024-03-24:
ok, no problem ... i will reflash when available

tpanajott, 2024-03-24:
The new release is now available. Make sure you download the firmware file from the beta-branch when reflashing the panels.

niedi74, 2024-03-25:
Version 0.1.20 as well removed te Manger adress fieled in my case .. ( US version)

it tried it to flash direct via the bin file from beta 

Than i go back to the original on , & test it over the latest beta Docker image to flasch .. same result 

i need to check why it work on my side, like to the others