NSPManager/NSPanelManager #142: No confirmantion of chosen file when upoading TFT-file

Ellghe, 2024-03-24:
It looks the same as before chosing file.

[image]

tpanajott, 2024-03-25:
That's strange. What version are you running? It works for me. Maybe @cablesandcoffee could also have a look?

[image]

Ellghe, 2024-03-25:
I can se that I was too tired and to ethusiastic to help when writing my observations. After double checking you are right. Suggest that you resize (increase) the text and/or make it black or white in order to make it easier to observe.

tpanajott, 2024-08-31:
This is now deemed fixed. If you think otherwise, please reopen issue.

closed 2024-08-31 by tpanajott