NSPManager/NSPanelManager #144: [BUG][BETA] On reboot current status on lights does not show on Panel

cablesandcoffee, 2024-04-06:
Setup:
Raspberry Pi 4
Docker aarch64
Openhab

Behavior:
On boot of panel it does not get current light values from manager so the screen says all lights are off all though they are on.

I've tested saving manager settings and when doing that all panels get and update the screen to the current light entities values so in that case it works.

tpanajott, 2024-04-07:
This should be fixed as of db8c5da7301d17d450ae024568fbc6d7a251d2db but will have to be confirmed with the next release

cablesandcoffee, 2024-04-12:
Seem to be working better. Brightness values get to the panel now. Kelvin is always 100 no matter what the settings are.

tpanajott, 2024-04-13:
Can you confirm in the logs from the MQTTManager (the container logs) or from MQTT that the correct kelvin is sent out or? That would indicate whether the error is in the manager or the panel firmware.

cablesandcoffee, 2024-04-13:
I have done tests again and I take back that brightness works. Don't know how I came to that conclusion.
Behavior right now:

Allt lights are OFF in Openhab
If I restart the manager the values that are seen in the MQTT log is:

nspanel/entities/light/22/state_brightness_pct 0
nspanel/entities/light/22/state_kelvin 48
nspanel/entities/light/49/state_brightness_pct 0
nspanel/entities/light/49/state_brightness_pct 72
nspanel/entities/light/49/state_hue 35
nspanel/entities/light/49/state_sat 72
nspanel/entities/light/49/state_kelvin 48
nspanel/entities/light/48/state_brightness_pct 0
nspanel/entities/light/48/state_brightness_pct 72
nspanel/entities/light/48/state_hue 35
nspanel/entities/light/48/state_sat 72
nspanel/entities/light/48/state_kelvin 48
nspanel/entities/light/31/state_brightness_pct 0
nspanel/entities/light/31/state_kelvin 48
nspanel/entities/light/33/state_brightness_pct 0
nspanel/entities/light/19/state_brightness_pct 0
nspanel/entities/light/19/state_brightness_pct 70
nspanel/entities/light/19/state_hue 36
nspanel/entities/light/19/state_sat 70
nspanel/entities/light/19/state_kelvin 46
nspanel/entities/light/51/state_brightness_pct 0
nspanel/entities/light/52/state_brightness_pct 0
nspanel/entities/light/27/state_brightness_pct 0
nspanel/entities/light/42/state_brightness_pct 0
nspanel/entities/light/42/state_kelvin 48
nspanel/entities/light/25/state_brightness_pct 0
nspanel/entities/light/25/state_kelvin 46
nspanel/entities/light/32/state_brightness_pct 0
nspanel/entities/light/32/state_kelvin 48
nspanel/entities/light/45/state_brightness_pct 0
nspanel/entities/light/45/state_brightness_pct 66
nspanel/entities/light/45/state_hue 37
nspanel/entities/light/45/state_sat 66
nspanel/entities/light/45/state_kelvin 48
nspanel/entities/light/34/state_brightness_pct 0
nspanel/entities/light/34/state_kelvin 48
nspanel/entities/light/21/state_brightness_pct 0
nspanel/entities/light/21/state_kelvin 48
nspanel/entities/light/36/state_brightness_pct 0
nspanel/entities/light/36/state_kelvin 48

If I after this restart a panel the panel updates four lights to on, the four lights above that have pct values above 0, not correct values though as discussed in other bug report. Kelvin slider is 100 in all rooms. This is somewhat correct since four ligts actually as values above 0. But the behavior below in different scenario is much weirder:

If I turn on all lights with the panel, check that all lights are ON in openhab and then restart the panel 
the panel sets five random lights as ON.  Really hard to see a pattern of why it

tpanajott, 2024-04-14:
Is this not the same issue as #145 ?

cablesandcoffee, 2024-04-14:
No not exactly I think. This i when you reboot a panel. #145 is when you restart the manager and the manager pushes out current light settings to the panels.

cablesandcoffee, 2024-04-14:
But this works now, I've tested with Openhab. There is only one minor thing left I think. It looks like Switch lights (ON/OFF) has an impact on the brightness slider. 

Example:
All lights are OFF
Set lights in a room to 70% (in the room there are 5 dimmer lights and 1 switch lights)
Lights turn on
Restart panel
Panel will show a brightness value of 76% in the room, Im guessing maybe because it calculates 100% for the switch light? Maybe. But Im thinking it should say 70% just as it did before rebooting the screen.

tpanajott, 2024-04-14:
Yes, a switch items that is on is considered 100% brightness.

cablesandcoffee, 2024-04-14:
Oh sorry it does not work with kelvin still. Is puts all room kelvin to 100%

About the switch brightness. When you turn lights on in a room i does not seem to take in to account about the switch.
Its only on reboot that the number increases. It feels wrong to have the switch light in the calculation.

tpanajott, 2024-04-14:
It does not update the screen with new values after you've pressed it to avoid flicker, it's still calculated the same though. If you change on another panel all other panels would show the wrong values.

cablesandcoffee, 2024-04-14:
Oh yeah thats true, saw that the same panel updated to that value as well when I scrolled through the rooms. Well, then that is as expected then. 

But back to kelvin. Both reboot of panel and restart of manager results in panel setting 100% kelvin everywhere.

tpanajott, 2024-04-14:
Which is very strange and I don't have an answer for that. The correct Kelvin values are sent out over mqtt but it seems they aren't registered on the light entities on the panel or they are not calculated correctly.

cablesandcoffee, 2024-04-14:
Okay Im closing this one now, when I test right now to reboot a panel. Everything works as expected.
Will continue to write in the other issue about restart of manager.

closed 2024-04-14 by cablesandcoffee