NSPManager/NSPanelManager #145: [BUG][BETA] Wrong states when panel gets current lights states

cablesandcoffee, 2024-04-06:
Setup:
Raspberry Pi 4
Docker aarch64
Version: 0.1.35-public_beta_hotfix3
Openhab

Expected behavior
On restart of manager all panels get updates with current light states. Both brightness and kelvin gets updated.

Current behavior:
Tested on my Kitchen page where I have three lights.
All three lights are set to 8% brightness and 25 kelvin

When I restart the manager this is seen in MQTT log:
nspanel/Kitchen/command {"command":"reload"}
nspanel/entities/light/1/state_brightness_pct 8
nspanel/entities/light/1/state_kelvin 25
nspanel/entities/light/2/state_brightness_pct 8
nspanel/entities/light/2/state_kelvin 25
nspanel/entities/light/3/state_brightness_pct 8
nspanel/entities/light/3/state_kelvin 25

States are correct but panels Kitchen page update with other value.
Brightness in this case is set to 20%
Kelvin is always 100%

tpanajott, 2024-04-07:
I need more information on all the lights configured in the kitchen and what capabilities they have.

cablesandcoffee, 2024-04-07:
I have 3 lights in Kitchen.
All dimmable. 
2 Brightness and colortemp
1 Brightness, colortemp and color

tpanajott, 2024-04-13:
Could you retry this but with the setting "MQTT ignore time (ms)" set to "0" and see if it changes. For this to actually works, do the following:
1. Change the setting to 0.
2. Reboot the NSPanel.
3. Once it has restarted, restart the MQTTManager from the first page "Actions"-menu.
4. Inspect result.

cablesandcoffee, 2024-04-13:
okay I have done some more tests. I would say that the brightness actually works. The problem is the rooms which have RGB lights. The saturation value on all RGB lights becomes the brightness value and then it becomes confusing when m trying to see if the values are correct. For example I put all lights to 100 % in a room. Restart manager, and the brightness drops to 80. If I change the saturation on all lights to 100 then the entire slider says 100 when I restart the manager.

tpanajott, 2024-04-13:
> okay I have done some more tests. I would say that the brightness actually works. The problem is the rooms which have RGB lights. The saturation value on all RGB lights becomes the brightness value [...]

That issue should be solved in the latest dev release

tpanajott, 2024-04-14:
Please try with latest dev release and have a look at the kelvin debug output from the NSPanel in the log. This should indicate whether any of the variables are to small and what averages it actually calculates.

cablesandcoffee, 2024-04-14:
Still shows all rooms as 100%kelvin  
Log:

19:45:18	DEBUG	HomePage.cpp:606 f(.updateLightStatus) Total kelvin: 100
19:45:18	DEBUG	HomePage.cpp:570 f(.updateLightStatus) Total num table lights: 4
19:45:18	DEBUG	HomePage.cpp:569 f(.updateLightStatus) Total kelvin table lights: 400
19:45:18	DEBUG	HomePage.cpp:543 f(.updateLightStatus) Total num table lights: 1
19:45:18	DEBUG	HomePage.cpp:542 f(.updateLightStatus) Total kelvin table lights: 100
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/52/state_brightness_pct
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/51/state_brightness_pct
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/49/state_sat
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/49/state_hue
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/49/state_kelvin
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/49/state_brightness_pct
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/48/state_sat
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/48/state_hue
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/48/state_kelvin
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/48/state_brightness_pct
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/45/state_sat
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/45/state_hue
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/45/state_kelvin
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/45/state_brightness_pct
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/42/state_kelvin
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/42/state_brightness_pct
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/36/state_kelvin
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/36/state_brightness_pct
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/34/state_kelvin
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/34/state_brightness_pct
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/33/state_brightness_pct
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/32/state_kelvin
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/32/state_brightness_pct
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/31/state_kelvin
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/31/state_brightness_pct
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/27/state_brightness_pct
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/25/state_kelvin
19:45:18	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/25/state_brightness_pct
19:45:17	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/22/state_kelvin
19:45:17	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/22/state_brightness_pct
19:45:17	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/21/state_kelvin
19:45:17	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/21/state_brightness_pct
19:45:17	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/19/state_sat
19:45:17	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/19/state_hue
19:45:17	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/19/state_kelvin
19:45:17	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/entities/light/19/state_brightness_pct
19:45:17	DEBUG	HomePage.cpp:570 f(.updateLightStatus) Total num table lights: 4
19:45:17	DEBUG	HomePage.cpp:569 f(.updateLightStatus) Total kelvin table lights: 400
19:45:17	DEBUG	HomePage.cpp:543 f(.updateLightStatus) Total num table lights: 1
19:45:17	DEBUG	HomePage.cpp:542 f(.updateLightStatus) Total kelvin table lights: 100
19:45:17	DEBUG	ScreensaverPage.cpp:127 f(.weatherMqttCallback) Received new weather data.
19:45:17	DEBUG	MqttManager.cpp:129 f(._mqttClientCallback) nspanel/status/weather
19:45:17	DEBUG	HomePage.cpp:606 f(.updateLightStatus) Total kelvin: 100
19:45:17	DEBUG	HomePage.cpp:570 f(.updateLightStatus) Total num table lights: 4
19:45:17	DEBUG	HomePage.cpp:569 f(.updateLightStatus) Total kelvin table lights: 400
19:45:17	DEBUG	HomePage.cpp:543 f(.updateLightStatus) Total num table lights: 1
19:45:17	DEBUG	HomePage.cpp:542 f(.updateLightStatus) Total kelvin table lights: 100
19:45:17	DEBUG	HomePage.cpp:606 f(.updateLightStatus) Total kelvin: 100
19:45:17	DEBUG	HomePage.cpp:570 f(.updateLightStatus) Total num table lights: 4
19:45:17	DEBUG	HomePage.cpp:569 f(.updateLightStatus) Total kelvin table lights: 400
19:45:17	DEBUG	HomePage.cpp:543 f(.updateLightStatus) Total num table lights: 1
19:45:17	DEBUG	HomePage.cpp:542 f(.updateLightStatus) Total kelvin table lights: 100
19:45:17	DEBUG	RoomManager.cpp:194 f(.loadAllRooms) Calling roomChangedCallback
[... cut ...]
19:45:17	DEBUG	RoomManager.cpp:344 f(.loadRoom) Removing scene: Movie Scene. ID: 8
19:45:17	INFO	RoomManager.cpp:182 f(.loadAllRooms) Getting config for room 8
19:45:17	DEBUG	RoomManager.cpp:344 f(.loadRoom) Removing scene: Scen Brasa. ID: 17
19:45:17	DEBUG	RoomManager.cpp:344 f(.loadRoom) Removing scene: Maten är klar. ID: 13
19:45:17	DEBUG	RoomManager.cpp:344 f(.loadRoom) Removing scene: Laga mat. ID: 12
19:45:17	INFO	RoomManager.cpp:182 f(.loadAllRooms) Getting config for room 2
19:45:17	DEBUG	RoomManager.cpp:29 f(.performConfigReload) Got reload command, reloading rooms.

tpanajott, 2024-04-14:
It seems to consider each individual light att 100% Kelvin for some reason.

cablesandcoffee, 2024-04-14:
Yes it seems like it. :)

tpanajott, 2024-04-14:
Have you set the values for min/max Kelvin correctly?

cablesandcoffee, 2024-04-14:
Hmm good question. It was set to 2000 and 6000. But I changed it to 2000 and 6535 instead. I would say that it works great now. :) Don't rally understand how it could make such a big difference. If lights are off, does there kelvin have an effect on the overall kelvin for that room?

tpanajott, 2024-04-14:
No, if the lights are off they are ignored.

cablesandcoffee, 2024-04-14:
Man Im mixing these two different issues...
On restart of manager it still sets lights to 100%. Restar of panel seems to work now though.

cablesandcoffee, 2024-04-14:
I changed to issue name now cause it was so confusing and closed the other one.
Kelvin is considered 100% on all lights all though it's not. :)

tpanajott, 2024-04-15:
Finally managed to find it. It seems I've managed to fix it. Please try again and report back.

tpanajott, 2024-04-15:
Bug confirmed fixed, will close.

closed 2024-04-15 by tpanajott