NSPManager/NSPanelManager #148: screensaver_mode -  no weather data on screensaver_mode change if panel setting is datetime

cablesandcoffee, 2024-04-13:
**Describe the bug**
No weather data when user changes screensaver_mode from bootup setting "datetime" to "weather"

**To Reproduce**
Steps to reproduce the behavior:
1. Set panel screensaver_mode in manager to one of the "datetime" settings and reboot panel
2. Wait for screensaver to start. The no weather screen saver is shown
3. Send command to screensaver_mode topic and change to one of the weather choices
4. Screensaver changes but weather is never updated.

**Expected behavior**
Weather should be updated

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment info (please complete the following information):**
 - Openhab
 - NSPanelManager version [ 0.1.36-public_beta ]
 - NSPanel Firmware version [0.1.40]

cablesandcoffee, 2024-04-13:
Realize that this is probably the same problem as #146

tpanajott, 2024-04-13:
Is the "screensaver.curIcon" (and all the other weather items) on the screensaver page set? Are they global? The screensaver should be update even if that is not the current page on dispay.

tpanajott, 2024-04-13:
I think the issue is that the variables/items used on the screensaver page in the TFT are not accessible when no displaying that page as if you change to a page with the items it updates them on the next data from MQTT. Is this something that is possible to fix in the TFT or does it have to be updated through a workaround in the firmware? The best would be for it to be solved in the TFT as it would eat up a bit of memory on the panel if solved in the firmware.

cablesandcoffee, 2024-04-13:
All items on page "screensaver" and "screensaver2" are global so you should be able to set them before entering the page? Are you updating both sceensaver and screensaver2 items independent of which page the user has decided to use via the Screensaver Mode setting?

tpanajott, 2024-04-14:
Issue fixed.

closed 2024-04-14 by tpanajott