NSPManager/NSPanelManager #152: HUE slider on light page does not work above 255

cablesandcoffee, 2024-04-28:
**Describe the bug**
HUE slider on the individual lights page seems to only work between values 0-255

**To Reproduce**
Steps to reproduce the behavior:
1. Enter individual lights page on panel for RGB lights
2. Enter RGB mode
3. Use RGB slider
4. Values are always sent to light but when user presses above 255 the slider jumps down to like 40
5. Same behavior if sending commands from Openhab. Slider update properly but not when getting values above 255.

**Expected behavior**
Slider should be able to set values from 0-360

**Environment info (please complete the following information):**
 - Openhab
 - NSPanelManager version [0.1.39]
 - NSPanel Firmware version [0.1.91]

tpanajott, 2024-04-28:
This is now fixed in latest commit in beta branch. I've done some testing but I only own one RGB light. Please test some more and confirm that it is indeed fixed and then close the issue if resolved.

cablesandcoffee, 2024-04-28:
Works like a charm! Closing

closed 2024-04-28 by cablesandcoffee