NSPManager/NSPanelManager #154: Docker Compose Help

cloudbr34k84, 2024-05-01:
Does anyone have a docker-compose file i could use? Im ok with these, but i can't get this to work. it says it doesnt like property url or  homeassistant_api..

```services:
  nspanelmanager:
    image: docker.io/nspanelmanager/nspanelmanager-amd64
    container_name: nspanelmanager
    network_mode: host
    url: "https://nspanelmanager.com"
    webui: http://xxx.xxx.xx.xxx:8000/
    panel_icon: mdi:fit-to-screen
    ports:
     - 8000:8000
     - 8001:8001
    environment:
      IS_HOME_ASSISTANT_ADDON: "true"
    init: false
    homeassistant_api: true
    restart: unless-stopped
    labels:      
      - "traefik.enable=true"
      - "traefik.docker.network=traefik"
      - "traefik.http.routers.XXXX.tls=true"
      - "traefik.http.routers.XXXX.tls.certresolver=dnsresolver"
      - "traefik.http.routers.XXXX.rule=Host(`XXXX.xxxxxxxxx.xyz`)"
      - "traefik.http.routers.XXXX.entrypoints=https"
      - "traefik.http.routers.XXXX.service=svcXXXX"
      - "traefik.http.services.svcXXXX.loadbalancer.server.port=XXXX"
      - "traefik.http.routers.XXXX.middlewares=auth@file"
      - "com.centurylinklabs.watchtower.enable=true"
    networks:
      - default```

aaronouthier, 2024-05-01:
I am by no means an expert in compose, but I believe you're suppose to create a Home Assistant api key and specify it under the appropriate section. "True" is not a valid key...

Also, I believe the first line should be "services:" (without quotes), unindented

cloudbr34k84, 2024-05-01:
> I am by no means an expert in compose, but I believe you're suppose to create a Home Assistant api key and specify it under the appropriate section. "True" is not a valid key...
> 
> Also, I believe the first line should be "services:" (without quotes), unindented

How did you load this up?

i have no issue using this with other dockers as my template
When i try load this by via powershell it says
**services.nspanelmanager Additional property homeassistant_api is not allowed**

aaronouthier, 2024-05-01:
I haven't kept up with NSPanel Manager, as my NSPanels are both disconnected at the moment, pending a certified electrician. Apparently California doesn't like me wiring my own home...

When I last used it, it was with a custom-made docker-compose.yml file. The one the devs supplied didn't work for me, but a bit of trial and error (and some motrin...) got it going for me.

cloudbr34k84, 2024-05-01:
is this project very active

aaronouthier, 2024-05-01:
Also, you state you're using powershell. Is this in Windows or Linux? Home Assistant doesn't run on Linux, so in that case, your install wouldn't be an add-on. Add-ons are for the OS variant only, must run on the same box, and be installed directly via HAOS.

aaronouthier, 2024-05-01:
> is this project very active

Yes! But the devs are in Great Brittan. They're likely asleep.

Check the completed issues to see how active it is.

cloudbr34k84, 2024-05-01:
> Also, you state you're using powershell. Is this in Windows or Linux? Home Assistant doesn't run on Windows, so in that case, your install wouldn't be an add-on. Add-ons are for the OS variant only, must run on the same box, and be installed directly via HAOS.
isnt this a docker of sorts, like any other docker? see this i smy lack of IT knowledge lol

aaronouthier, 2024-05-01:
While HAOS does run docker behind the scenes, it does it in a certain way. Even running the container version of HA doesn't even have the capability to run add-ons, as the menu option is missing.

aaronouthier, 2024-05-01:
Most of the issues were raised by the 2 devs: tpanajot and cablesandcoffee. I hope I spelled those correctly!

cloudbr34k84, 2024-05-01:
im reading the manual and it says for installation its using a standaloine docker and or the HA Add on? thats how i read it

2.1.2 Standalone Docker container
2.1.3 Home Assistant add-on

aaronouthier, 2024-05-01:
Looking again at the source, I see the devs have an updated compose.yml reference, which it seems you used. The devs should know what they are doing there, but I had issues with their old one, so 🤷🏻‍♂️.

Try temporarily commenting out the offending lines, and see what it gives you?

aaronouthier, 2024-05-01:
If using as an add-on, specify TRUE, if not, specify false. If it's running on a different machine or different VM, then it is not considered an add-on.

aaronouthier, 2024-05-01:
Also, just a hunch here, but try specifying the url: field without quotes. The webui field doesn't have quotes, and it doesn't sound like it complains on that one.

cloudbr34k84, 2024-05-01:
for now i have just gone down the path of the HA Add rather than a stand alone docker.  I just wanted to check out this project and see if wnat to invest in a panel

Man thanks for your help though!!

closed 2024-05-01 by cloudbr34k84

aaronouthier, 2024-05-01:
I believe the OP was referencing this file:
https://github.com/NSPManager/NSPanelManager/blob/main/docker-beta/config.yaml

tpanajott, 2024-05-01:
> I believe the OP was referencing this file:
> https://github.com/NSPManager/NSPanelManager/blob/main/docker-beta/config.yaml

Probably but that file is so that HA knows how to run the addon and start the container in the background.

cloudbr34k84, 2024-05-01:
ahh yes! i got my docker files mixed up!