NSPManager/NSPanelManager #155: Add keypad screen/pop-up for alarm integration

EvilMilkshake, 2024-05-01:
**Describe the solution you'd like**
A clear and concise description of what you want to happen.

It would be great to have the ability to arm/disarm or set state home/away from a keypad to integrate with HA and Alarmo. [https://github.com/nielsfaber/alarmo](url)

cablesandcoffee, 2024-11-22:
[image]
Suggestion as alarm page, triggered from Alarm symbol on dropdown page.
Colored icon is current state. User presses the desired state, change color and gets to insert pin code.

EvilMilkshake, 2024-11-23:
Looks great! I think you could put the state buttons on the top for the US portrait version.

Excited for this.