NSPManager/NSPanelManager #156: NSPanel not connecting to wifi at all

PixyDon, 2024-06-22:
Hey Guys,

I've followed the Technical Setup installing NSPanel as the addon on Home Assistant and consistently becoming stuck at the wifi configuration section.  I've watched the Cables & Coffee setup video ans well as the Mark Watt Tech video and still the issue persists.  

Steps to reproduce the behavior:
1. I get to the point where I access the NSPanel wifi access point.
2. Add my HA ip addess as the NSPanel Manager address.
3. Choose from list my SSID and add password.
4. Add MQTT user and password details and click Save.
5. I can see on my wifi router the panel has been picked up.
6. Panel's wifi symbol appears disconnected, doesn't appear in NSPanel Manager.
7. Re-flash and trt again over and over, same result.

 - Home Assistant
 - Chrome
 - 0.1.2.3
 - 0.1.2.5

tpanajott, 2024-06-22:
What WiFi för you have? Specifically wifi-secutity? We had a user on the discord a few weeks ago that had issues with his WPA3 security as its not really supported. Switching to WPA2 solved it.

PixyDon, 2024-06-22:
> What WiFi för you have? Specifically wifi-secutity? We had a user on the discord a few weeks ago that had issues with his WPA3 security as its not really supported. Switching to WPA2 solved it.

[image]

Seems like split between the 2, I'll see if changing it to WPA2 only helps...

PixyDon, 2024-06-22:
OK, I'm not sure it has actually do something.  But I now have the following issue, even though adding a room and lights to a room brings up the fill list of entities??

[image]

tpanajott, 2024-06-22:
That message comes from the MQTTManager backend component. It says it can't connect to MQTT for whatever reason (have a look in the logs for the addon/container). The list of entities when populating rooms have nothing to do with MQTT. They are gathered directory from Home Assistant and/or OpenHAB. So, your HA connection is working but not MQTT.

PixyDon, 2024-06-22:
> That message comes from the MQTTManager backend component. It says it can't connect to MQTT for whatever reason (have a look in the logs for the addon/container). The list of entities when [...]

Thanks, I hadn't created an MQTT User and had used the default homeassistant usr/pwd.

Added a user and re-flashed, went through setup again and it seems to be working now.  Just need to get the screen updated...

closed 2024-06-23 by PixyDon