NSPManager/NSPanelManager #158: Fails to get weather when running nspanelmanger on Raspberry Pi

cablesandcoffee, 2024-07-23:
**Describe the bug**
When running nspanelmanager from a pre-packaged docker image on a raspberry Pi the weather data always fails to get fetched.

Strangely it works when starting the container from the docker-build_and_run_dev.sh -script.

Nspanelmanger log:

[2024-07-22 22:06:30.429] [error] [WebHelper.cpp:65] [35] Failed to perform HTTP request to 'https://api.open-meteo.com/v1/forecast?latitude=65.31717&longitude=21.47944&current=temperature_2m,relative_humidity_2m,precipitation,weather_code,wind_speed_10m,wind_gusts_10m&daily=weather_code,temperature_2m_max,temperature_2m_min,sunrise,sunset,precipitation_sum,precipitation_probability_max,wind_speed_10m_max,wind_gusts_10m_max&timeformat=unixtime&wind_speed_unit=ms&timezone=Europe/Stockholm&precipitation_unit=mm&temperature_unit=celsius'. Got response data: ''
[2024-07-22 22:06:30.429] [error] [WebHelper.cpp:66] [35] curl_easy_perform() failed, got code: 60. Text interpretation: SSL peer certificate or SSH remote key was not OK
[2024-07-22 22:06:30.429] [error] [WebHelper.cpp:67] [35] Curl error buffer: SSL certificate problem: unable to get local issuer certificate
[2024-07-22 22:06:30.429] [error] [weather.cpp:71] [35] Failed to get weather.

**Environment info (please complete the following information):**
 - NSPanelManager version [0.1.41-public_beta3]

tpanajott, 2024-07-24:
I will have to take a look at this. The strange thing is that running the container in -dev mode works. So, something is installed during dev mode that makes it work is my guess. But again, it works for other architectures so a very strange problem.

I have a raspberry pi at home and will have to take a look at it with that one.

tpanajott, 2024-07-25:
Dang it, forgot there was a registered issue for this bug. Anyway, it's been solved by commit 33b410fe49cd7a6f6d16d4791312b8b7555dae4d. I'm closing this issue.

closed 2024-07-25 by tpanajott