NSPManager/NSPanelManager #170: Update tft file weather page default values

cablesandcoffee, 2024-08-30:
Right now there are random values and icons on the weather page left from building and testing the page in Nextion editor.
Remove random numbers in nextion editor and set same default icons everywhere so it looks clean if the weather does not load or takes a couple of seconds to load.

tpanajott, 2024-08-30:
Perhaps it would be a good idea to somehow indicate that data is missing? A special icon? Perhaps default temperature is 255 or something totalt unreasonable?

cablesandcoffee, 2024-08-30:
Yes I have a plan! m thinking just a hyphen for the numbers which indicated data is missing. And for the icons we have a supernice weather icon with an exclamation mark! :)

cablesandcoffee, 2024-08-30:
[image]

cablesandcoffee, 2024-08-31:
Fixed. Closing

closed 2024-08-31 by cablesandcoffee