NSPManager/NSPanelManager #171: Light is already off, vill not send update to Openhab (All though light is ON, happens after a light has been set to a color)

cablesandcoffee, 2024-08-31:
**Describe the bug**
After setting light to a color (HSB) light cannot be turned off. 

**To Reproduce**
Steps to reproduce the behavior:
1. Enter room page with a color light
2. Click on the color light to enter individual light page
3. change to color mode and set a color on HUE slider. 
4. go back to room page
5. toggle off the color light that we just changed to a color (light turns OFF)
6.  toggle ON the same light (light turns ON) Log in manager says" Setting light 45:Måne to HSB: 360, 100, 100)
7. Toggle OFF light again (Light does not turn off, log in manager says: "Light is already off, vill not send update to Openhab"

I think it keeps behaving like that until you you use the color temp slider and then change the brightness on main page. After that you can turn the light OFF again.

**Environment info (please complete the following information):**
 - Openhab
 - NSPanelManager version [ 0.1.43-public_beta_h3 ]
 - NSPanel Firmware version [0.1.149]

**Additional context**
Add any other context about the problem here.

tpanajott, 2024-08-31:
When sending a command to turn on the light with RGB the proper flag for saying the list is on was never set. This has been fixed now, please verify and close if fixed.

cablesandcoffee, 2024-08-31:
Fixed! Closing!

closed 2024-08-31 by cablesandcoffee