NSPManager/NSPanelManager #172: Weather forecast sometimes show night icon

cablesandcoffee, 2024-08-31:
**Describe the bug**
Weather forecast seems to use night icons sometimes

**Expected behavior**
Night icons should not be used on the 5 day forecast? That's what I'm thinking anyway. :)

Thoughts:
Right now it seems to use night icons on forecast a bit randomly. 
Right now on my panel I have:
Saturday : cloud and sun
Wednesday: moon and a cloud. 
It's the same weather but different icons are used.

tpanajott, 2024-08-31:
Hm, that doesn't sound right. It's not something I've seen, or at least noticed. Should be an easy fix if we simply decide that daily forecasts should never use night icons?

cablesandcoffee, 2024-08-31:
Yes I'm thinking so as well. That looks best I think! :)

tpanajott, 2024-08-31:
Disabled night icons for forecast. Please verify and close if deemed fixed.

cablesandcoffee, 2024-08-31:
Hmm, still see night icons on forecast.

tpanajott, 2024-08-31:
That's shouldn't be possible. Is the icon mapping from day/night correct?

cablesandcoffee, 2024-09-09:
I will check the mapping! Could be a fault there as you say. Didn't think of that. :)

cablesandcoffee, 2024-09-14:
Checked the mapping but it looks correct:

    {
      "id": "0",
      "character-mapping-day": "A",
      "character-mapping-night": "B",
      "text": "Clear sky"
    },
    {
      "id": "1",
      "character-mapping-day": "C",
      "character-mapping-night": "D",
      "text": "Mainly clear"
    },
    {
      "id": "2",
      "character-mapping-day": "C",
      "character-mapping-night": "D",
      "text": "Partly cloudy"
    },
    
C  is "cloud with sun" icon and D is the "cloud with moon" icon.

I did make some changes to the file anyway since i realized "mainly clear" should probably use A/B as well but that does not have to do with this problem.

tpanajott, 2024-09-14:
hm, very strange. Perhaps we have to take a look at the actual data being sent from OpenMeteo. Could be something with that specific mapping?

cablesandcoffee, 2024-09-15:
But I don't think OpenMeteo says anything about day/night, it's just "mainly clear" for example and for mainly clear we have two different icons to use. The logic of using the day/night icons is in the manager right? 

Test right now at 08:28AM:
When Råsunda is set the panel is getting the D icon (night) - not correct
When Piteå is set the panel is getting the C icon (day) - correct

I haven't checked what data is coming from OpenMeteo but I'm thinking it must be in our own code that we choose to use the night icon in some cases? or am I missing something? :)

This is the data if I switch to Råsunda:

nspanel/status/weather {"forecast":[
{"day":"Sun","icon":"D","maxmin":"19°/8°","pre":"0mm","prepro":"0%","wind":"4m/s"},{"day":"Mon","icon":"J","maxmin":"18°/8°","pre":"0mm","prepro":"47%","wind":"3m/s"},{"day":"Tue","icon":"C","maxmin":"21°/11°","pre":"0mm","prepro":"14%","wind":"3m/s"},{"day":"Wed","icon":"D","maxmin":"20°/10°","pre":"0mm","prepro":"0%","wind":"2m/s"},{"day":"Thu","icon":"D","maxmin":"21°/9°","pre":"0mm","prepro":"0%","wind":"3m/s"}],"icon":"A","maxmin":"19°/8°","prepro":"0%","sunrise":"06:17","sunset":"19:08","temp":"9°","wind":"3m/s"}

This is the data if I switch to Piteå:

nspanel/status/weather {"forecast":[
{"day":"Sun","icon":"C","maxmin":"15°/5°","pre":"0mm","prepro":"0%","wind":"3m/s"},{"day":"Mon","icon":"C","maxmin":"17°/4°","pre":"0mm","prepro":"4%","wind":"5m/s"},{"day":"Tue","icon":"J","maxmin":"17°/10°","pre":"0mm","prepro":"24%","wind":"4m/s"},{"day":"Wed","icon":"C","maxmin":"18°/10°","pre":"0mm","prepro":"0%","wind":"4m/s"},{"day":"Thu","icon":"C","maxmin":"21°/11°","pre":"0mm","prepro":"8%","wind":"5m/s"}],"icon":"A","maxmin":"15°/5°","prepro":"0%","sunrise":"05:57","sunset":"19:01","temp":"6°","wind":"3m/s"}

tpanajott, 2024-09-15:
I had a second look in the code and it seems like the fix implemented earlier has somehow not made it into current beta/stable. Please test again with latest beta from GitHub.

cablesandcoffee, 2024-11-21:
Late answer byt this never happens anymore so it must be fixed!

closed 2024-11-21 by cablesandcoffee